madhephaestus/ESP32Encoder #2: Hardware

dpharris, 2019-07-18:
Can you add a section to the README about connecting encoders to the ESP32?  Does one connect the encoder common to +V or ground?  Are pull-up or pull-down resistors required?  

Thanks!

madhephaestus, 2019-07-18:
I'm not sure what you're suggesting here. The pull downs ensure that you do not get open-collector noise and ensure when no encoder is connected, or if the encoder breaks, then the library will not register any ticks. 

Common usually refers to ground, not V+.

dpharris, 2019-07-18:
Sorry for the confusion.

1.  Can you please add a small section in your README to tell users how to
hook up their encoders.

2.  Encoders have two internal switches connected to a common terminal.
Since you have defined pull-down on the inputs, then each switch will need
to pull-up the input when closed, and therefore the common-terminal of the
encoder needs to be connected to V+.
----------               -------------
ESP32 pinA+-------------+out1 Encoder |
      pinB+-------------+out2   common+----------o V+
----------               -------------

Is that right?

Thanks,
David

madhephaestus, 2019-07-18:
1) I avoid the responsibility for hookup instructions because there is *no* "right" way to do it. That information exists, and should stay, in the data sheet for any specific encoder. This module is expecting Drive high, drive low low impedance digital signals (normal microcontroller digital input stuff), and how to hook your specific device up to that is up to you and your devices datasheet. 

2) To my point in 1), that is not how the encoders i use work. They are photointerrupters, not switches. I have never know an encoder where you can safely reverse power and ground. Do you have a link to the datasheet for yours so i can investigate?

dpharris, 2019-07-18:
Many experimenters will use mechanical rotary encoders like:
https://www.bourns.com/docs/Product-Datasheets/PEC11L.pdf  The mechanical
decoders suggest a circuit with pull-ups on the signal-outs and their
common to ground.  However, since you activate the ESP32's internal
pull-downs, I can connect my common to V+.

You have a good point, one should not assume the hardware.

Would you be willing to add something like "The internal pull-downs are
activated on the selected encoder inputs. "

Thanks
David

madhephaestus, 2019-07-18:
ok, done!

closed 2019-07-18 by madhephaestus