madhephaestus/ESP32Encoder #8: erroneous count-up on sup period rotation using SingelEdge

404-baitnotfound, 2019-09-21:
when using attachSingelEdge() I get erroneous count-up by wiggling the encoder shaft.
Using attachHalfQuad I get correct half step counts.

This is most likely due to the incorrect edge triggering of the ESP32.  (but don't quote me on that)

To ignore singel count changes I wrote following code in my loop:

```
int32_t encoder_position = 0;

void loop() {
  
  uint16_t enc_delta = abs(encoder.getCountRaw() - (encoder_position*2)) / 2;
  if(encoder.getCountRaw() > (encoder_position*2)){
    encoder_position += enc_delta;
  }
  else{
    encoder_position -= enc_delta;
  }
}
```

madhephaestus, 2019-09-21:
no, thats expected behavior in single edge mode. Single edge mode is for counting button presses NOT quadrature style encoders.

closed 2019-09-21 by madhephaestus

404-baitnotfound, 2019-09-21:
ok how is single edge mode intended to be used with buttons?  It still requires two pins?

madhephaestus, 2019-09-21:
give the same pin value for both, or connect the pins together