madhephaestus/ESP32Encoder #12: Was working before update, now won't and now I can't figure it out.

TelegramSam, 2020-04-12:
I updated the ESP32 board definition and updated ESP32Encoder to the latest value. Now the encoder doesn't work right, showing 0 as a count or -1, but never counting up or down like I used to.

I've experimented with various pull-up and pull-down pin settings, and have had no improvement.

I'm using Pins 21 and 22 on an ESP32 Dev Module, with the center pin connected to ground. Similar to the setup used in this issue:

https://github.com/madhephaestus/ESP32Encoder/issues/11

That confuses me slightly, as this issue seems to indicate that the built in pull is _down_ not up: https://github.com/madhephaestus/ESP32Encoder/issues/7

Any ideas?

madhephaestus, 2020-04-12:
Can you switch the encoder version back to the one you had when it works and try again? I want to isolate the toolchain change from the encoder change. Let me know what version worked for you?

TelegramSam, 2020-04-12:
I'll try. I don't think it was the library version change, but the ESP32 changes. I did update the encoder library to see if that would help, but it didn't. 

I'll give it some testing, and see what I can find.

TelegramSam, 2020-04-13:
I found the problem: a broken wire connection to the rotary encoder. ::facepalm:: Tested with newest versions of everything and it all works. Sorry for the bother!

closed 2020-04-13 by TelegramSam