madhephaestus/ESP32Encoder #13: Counter Skipping

TelegramSam, 2020-04-18:
I've attached an KY040 rotary encoder to a knob that has a stop. (I've used printed gears on both the knob and the encoder.)

I want to track knob position with zero at the knob stop, but I can tell some skipping is occurring as the rotary encoder turns. To test the skip, I zero the counter at the stop, turn the knob a number of rotations, and turn it back to the stop. If all works right, the number should be 0 again. Here are some numbers I gathered in test, moving slow and fast, and changing encoder mode just to see if it would help:

```
at stop, middle value, back at stop.

half quad
0 620 -30 slowly
0 603 -90 fast

full quad
0 1204 -210 slowly
0 1222 -60 fast

single edge
0 1318 -141 slow (number jumped around)
0 1172 897 fast
```

Any ideas how I debug this? I don't think it's just speed, as the problem isn't significantly decreased at slow speeds. I don't have a scope to check the wire signals. The sketch is pretty simple, with one other interrupt that I'm careful with. No wifi or bluetooth being used.

TelegramSam, 2020-04-18:
Also, I have the center pin connected to ground.

madhephaestus, 2020-04-18:
There is no indication this is a code issue, i strongly suspect electrical issues. I think you need to get a scope to debug this. These are ugly but work well: https://www.amazon.com/dp/B00C5TLV3O

madhephaestus, 2020-04-18:
If you can verify the electrical signal is stable, then open a ticket on the code.

closed 2020-04-18 by madhephaestus