madhephaestus/ESP32Encoder #14: E (33) pcnt: pcnt_get_counter_value(132): PCNT UNIT ERROR

JAY-M-ERICSSON, 2020-04-19:
Hi

Trying out your library with a pretty simple code and the very common KEYES Rotary Encoder with a switch (KY-040) with a basic "ESP 32 Dev Module"

here is the test code:
```
#include <ESP32Encoder.h>
ESP32Encoder encoder;
const uint8_t encoderCLKPin = 13; // encoder CLK
const uint8_t encoderDTPin = 14; // encoder DT
int32_t encoderCptr  = 0L;

void setup() {
  Serial.begin(115200);
  ESP32Encoder::useInternalWeakPullResistors = true; // Enable the weak pull down resistors
  encoder.clearCount();
  encoder.attachHalfQuad(encoderDTPin,encoderCLKPin);  // Attach pins for use as encoder pins
}

void loop()
{
  int32_t currentEncoderCptr = encoder.getCount();
  if (currentEncoderCptr != encoderCptr) {
    if (currentEncoderCptr > encoderCptr) {
      Serial.print(F("tick+  = "));
    } else {
      Serial.print(F("tick- = "));
    }
    encoderCptr = currentEncoderCptr;
    Serial.println(currentEncoderCptr);
  }
}
```

I occasionally get  tick+ or tick- that is unwanted - that's probably because my module is a poor quality $0.50 component bought from the Far East but I also get at boot the following error message:

_E (33) pcnt: pcnt_counter_clear(158): PCNT UNIT ERROR_

I'm curious what this error is and how to get rid of it ?

madhephaestus, 2020-04-19:
If you want to use KY-040 you need to understand mechanical switch bouncing. I bet what you are getting is a bounce that takes longer then the runt filtering built in. You need to add a small capacitor to the switch and only move it SLOWLY. Mechanical encoders are only stable at VERY low speeds, like a knob that someone moves with their fingers. 

Check out this document: http://web.engr.oregonstate.edu/~traylor/ece473/lectures/debounce.pdf for more information. 

Closing issue because it it an electrical problem.

closed 2020-04-19 by madhephaestus

JAY-M-ERICSSON, 2020-04-19:
I understand what bounces are and why I would possibly get tick errors. that's fine
I'm still unclear on the error message I get 
**E (33) pcnt: pcnt_counter_clear(158): PCNT UNIT ERROR**
there are similar messages when I compile your example as well.

madhephaestus, 2020-04-19:
try removing your encoder from the circuit entirely. Verify with an o-scope (at least 10 Mega-samples/sec) that the lines are clean. If you still get the error after verifying the signals are perfectly clean, let me know.

 I can not reproduce the error on my hardware, so i suspect electrical issues.

madhephaestus, 2020-04-19:
Also, what is the reason for trying to clear the encoder before attaching? Try removing that as well.

JAY-M-ERICSSON, 2020-04-19:
> Also, what is the reason for trying to clear the encoder before attaching? Try removing that as well.
Good catch - that's because in your example you did it like that :) 

I switched the order of the commands
```
  encoder.attachHalfQuad(encoderDTPin,encoderCLKPin);  // Attach pins for use as encoder pins
  encoder.clearCount();
```
  and now the error message is gone.

--> suggest you fix this in your example as well

thx for the good work and sharing !

madhephaestus, 2020-04-19:
example updated in 0.3.2