madhephaestus/ESP32Encoder #16: Help about your example

tombo9999, 2020-05-05:
Hi Kevin, sorry to bother you.
I'm just an hobbyist, trying to understand how to control position of an [N20 Micro Encoder Gear Motor](https://www.aliexpress.com/item/32886430913.html?spm=a2g0s.9042311.0.0.27424c4dXqgkLq)
Can you help me understand if I can use your library and how?

I made a simple arduino code (just googling around and copying some code), but I have very unstable result. As of now I don't understand if it's a software or an hardware problem :/

Thanks a lot in advance, Luca (from Italy)

madhephaestus, 2020-05-05:
Yes, that sort of encoder is exactly what this library is for!

Per the documentation for the motor:

Red → motor power +
Black → coded power + ( 3.3-5V)
Yellow → signal feedback (7 signals per turn)
Green → signal feedback (7 signals per turn)
Blue → Encoder Power Supply -Ground
White→Motor power supply - 

Have you used an oscilloscope to verify the power rails and that the encoder is producing clean signals? Also you must only run it at 3.3v, if you run at 5v you need to step down the signals through a level shifter into the esp32.

madhephaestus, 2020-05-05:
Also, go through the pin usage guide I made for my students to see if you are running into any of the pin usage problems: https://github.com/WPIRoboticsEngineering/RobotInterfaceBoard#io-pin-counts

tombo9999, 2020-05-05:
Hi again.
I'm using pin 14 and 15 on Ai-Thinker ESP32-Cam (see enclosed Fritzing pdf), but I don't have an oscilloscope. And last but not least, I don't understand the concept of pull-up and pull-down resistor, sorry.
[Xpider_perfboard_ver-01_bb.pdf](https://github.com/madhephaestus/ESP32Encoder/files/4581481/Xpider_perfboard_ver-01_bb.pdf)

I modified your example, but I get the following error in Arduino Serial Monitor.
Surely I made some silly mistake ;)
_ESP32encoder Booting...
16:40:44.379 -> E (1425) pcnt: pcnt_get_counter_value(137): PCNT UNIT ERROR
16:40:44.379 -> E (1425) pcnt: pcnt_get_counter_value(137): PCNT UNIT ERROR
16:40:44.379 -> Encoder Start = 16229_

`#include <ESP32Encoder.h>

// ESP32Encoder encoder;
ESP32Encoder encoder2;

// timer and flag for example, not needed for encoders
unsigned long encoder2lastToggled;
bool encoder2Paused = false;

void setup(){
  
  Serial.begin(115200);
  Serial.println("ESP32encoder Booting...");
  // Enable the weak pull down resistors
  ESP32Encoder::useInternalWeakPullResistors=true;
  
  // set starting count value
  encoder2.setCount(37);

  Serial.println("Encoder Start = "+String((int32_t)encoder2.getCount()));

  // Attache pins for use as encoder pins
  // encoder.attachHalfQuad(12, 13);
  // Attache pins for use as encoder pins
  encoder2.attachHalfQuad(14, 15);
  
  // clear the encoder's raw count and set the tracked count to zero
  encoder2.clearCount();

  // set the lastToggle
  encoder2lastToggled = millis();
}

void loop(){
  // Loop and read the count
  Serial.println("Encoder count = "+String((int32_t)encoder2.getCount()));
  delay(100);

  // every 5 seconds toggle encoder 2
  if (millis() - encoder2lastToggled >= 5000) {
    if(encoder2Paused) {
      Serial.println("Resuming Encoder 2");
      encoder2.resumeCount();
    } else {
      Serial.println("Paused Encoder 2");
      encoder2.pauseCount();
    }

    encoder2Paused = !encoder2Paused;
    encoder2lastToggled = millis();
  }
}
`

madhephaestus, 2020-05-05:
do not us pin 15 as an input, it is a strapping pin for the core and has external hardware on that pin.
https://github.com/WPIRoboticsEngineering/RobotInterfaceBoard#output-only-pins

also remove 
```encoder2.setCount(37);```

unless you want to start at a specific count. 

If changing the pin doesn't work, you will need to get an o-scope and look at the signals. This would do fine: https://www.amazon.com/Quimat-Oscilloscope-BNC-Clip-Assembled-Finished/dp/B07QML4LJL/

madhephaestus, 2020-05-05:
Also move the initial read of the encoder down below the attach. I changed that in the example in the latest release.

tombo9999, 2020-05-05:
It works with pin 12 and 13!!! Tried 13 and 14 but didn't work at all.
I still have a problem, as pin 12 need to be disconnected when uploading new code :/
Maybe I was wrong choosing ESP32-Cam (only few pin available).
For my project I need to control the encoded motor, another DC motor(not encoded), a servo and the camera.
Really thanks for your help :)
My really simple code for testing purpose:
```

#include <ESP32Encoder.h>

// ESP32Encoder encoder;
ESP32Encoder encoder;

// timer and flag for example, not needed for encoders
unsigned long encoderlastToggled;
long interval = 500;

void setup(){  
  Serial.begin(115200);
  Serial.println("ESP32encoder Booting...");
  // Enable the weak pull down resistors
  ESP32Encoder::useInternalWeakPullResistors=true;

  // Attache pins for use as encoder pins
  encoder.attachHalfQuad(12, 13);

  // set the lastToggle
  encoderlastToggled = millis();
}

void loop(){
  // toggle encoder at interval
  if (millis() - encoderlastToggled >= interval) {
    Serial.println("Encoder count = "+String((int32_t)encoder.getCount()));
    encoderlastToggled = millis(); 
  }
}

```

closed 2020-05-05 by tombo9999