madhephaestus/ESP32Encoder #17: Question about using pullups instead of pulldowns

mfarabee, 2020-05-07:
I am using an open collector NPN rotary encoder that will require pullups instead of pulldowns. I want to be sure that this library will work properly. I have two questions:

1) I see that I can set the "useInternalWeakPullResistors" to false, but how do I use the internal pullup resistor. It looks like if I set the pin to  "INPUT_PULLUP" in my sketch, your attach method will override it. Will I need to use external pullups and forget trying to use the internal pullups or can I redefine the pin after initializing the encoder?
   #define A_PIN 16
   #define B_PIN 17
   ESP32Encoder::useInternalWeakPullResistors=false;
   encoder.attachFullQuad(A_PIN, B_PIN);
   pinMode(A_PIN, INPUT_PULLUP);
   pinMode(B_PIN, INPUT_PULLUP);

2) Since your code is based on a pulldown, will the pulse detection be reversed or will it work correctly using a pullups? I don't know if this has any ramifications.

It would be nice to be able to specify using internal pullup as an option.

Thanks,

mfarabee, 2020-05-14:
I was able to answer my own questions after some experimenting. I was able to use to code above and it worked for using the NPN open collector. Direction was not a problem.

closed 2020-05-14 by mfarabee

Fusseldieb, 2020-06-21:
> I was able to answer my own questions after some experimenting. I was able to use to code above and it worked for using the NPN open collector. Direction was not a problem.

Can you please elaborate?

madhephaestus, 2020-06-21:
Pull resistors are now explicitly Up, Down or none using an enum.

mfarabee, 2020-06-23:
To elaborate on the question I asked:
Previously I was not able to define the pins as using the internal pullup, only pulldown or none. I was building a lathe DRO and had an encoder that required a pullup. I was hoping to use the internal pullup. The way that I achieved this was to disable the pulldown resistor in the code, attach the encoder, then redefine the pins to use the internal pullups. 

With the new update, this is no longer necessary as I can specify pullup, pulldown or none! 
THANK YOU!

madhephaestus, 2020-06-23:
:D

Fusseldieb, 2020-06-23:
> With the new update, this is no longer necessary as I can specify pullup, pulldown or none!

Can you please post a snipped of that code?
The documentation of this repo lacks some examples.

madhephaestus, 2020-06-23:
Per Arduino convention, the examples are in the examples folder: https://github.com/madhephaestus/ESP32Encoder/blob/master/examples/Encoder/Encoder.ino and are availible in the IDE like any other example program.