madhephaestus/ESP32Encoder #20: Library application for very fast encoders

fabiocassidori, 2020-05-26:
I'm going to use it on a very fast motor (12000rpm) and the encoder has a 12 bit resolution, therefore I will have an interrupt in about 1us. Do you think it will be possible to run a 500 lines of code with your library? 

**MADHELPHAESTUS Answer:**

this lib doesn't use pin change interrupts, it uses the PCNT module which is a hardware counting peripheral, so all you need to be concerned with is that the runt pulse filter is short enough so as not to filter your counts as runts.

https://github.com/madhephaestus/ESP32Encoder/blob/master/src/ESP32Encoder.cpp#L154

250 there means 250 clock cycles must pass between edge transitions to be detected as a count pulse. Do the math based on the RPM to get your ticks per second, then compute the smallest seconds per tick, and compute how many clock cycles at 240mhz that is. If your number is less that 50% more than the runt time, then it will work. If it comes in less than that number, you can try to change that line of code in your copy of the library to see if you can make it work. 

Also as a note, there are interrupts, but they only fire when a 16bit count buffer overflows.  

Short answer, its unlikely that you have an encoder fast enough to cause the PCNT module issues, but it does have known limits.

fabiocassidori, 2020-05-26:
Let I see if I understand.
My encoder will have about 60000 ticks per second, therefore about 1,67e-5 seconds per tick. That is about 4000 clock cicles (at 240mhz) per tick.  So, do I need to change the number "250" to something around "3000"? (4000 cycles is less that 50% more than "3000")

madhephaestus, 2020-05-27:
no that means you are good! this lib should work fine for you.

fabiocassidori, 2020-05-27:
Thank you very much, Kevin.
All the best.

closed 2020-06-08 by madhephaestus