madhephaestus/ESP32Encoder #27: Arduino 1.8.13 on ESP-32S-PICO-KIT V4.1 fails

myglc2, 2020-07-24:
COMMIT:
git log | head
commit a6f47f8d7014305d46010d2a5321d1d1323d8f77
Merge: 3752002 aa647e5
Author: Kevin Harrington <mad.hephaestus@gmail.com>
Date:   Thu Jul 23 18:05:09 2020 -0700

    Merge pull request #26 from thijstriemstra/patch-2

    fix typos

SERIAL MONITOR:

⸮$⸮⸮⸮⸮⸮⸮⸮b9⸮   : 0x00000000  LCOUNT  : 0x00000000  

Backtrace: 0x400e8db4:0x3ffb1ee0 0x400d1068:0x3ffb1f10 0x400d1156:0x3ffb1f40 0x400d0da9:0x3ffb1f60 0x400d1a53:0x3ffb1fb0 0x40088299:0x3ffb1fd0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5816
entry 0x400806ac
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x400e8db0: ffffffff ffffffff ffffffff
Core 1 register dump:
PC      : 0x400e8db4  PS      : 0x00060f30  A0      : 0x800d106b  A1      : 0x3ffb1ee0  
A2      : 0x00000011  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x3ffc0170  
A6      : 0x00000013  A7      : 0x00000000  A8      : 0x800e7ae9  A9      : 0x00000001  
A10     : 0x00000011  A11     : 0x3ffb8058  A12     : 0x00000001  A13     : 0x00000001  
A14     : 0x00060320  A15     : 0x00000000  SAR     : 0x00000010  EXCCAUSE: 0x00000000  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

Backtrace: 0x400e8db4:0x3ffb1ee0 0x400d1068:0x3ffb1f10 0x400d1156:0x3ffb1f40 0x400d0da9:0x3ffb1f60 0x400d1a53:0x3ffb1fb0 0x40088299:0x3ffb1fd0

[...]

myglc2, 2020-07-24:
ARDUINO UPLOAD LOG:

Sketch uses 231389 bytes (17%) of program storage space. Maximum is 1310720 bytes.
Global variables use 15596 bytes (4%) of dynamic memory, leaving 312084 bytes for local variables. Maximum is 327680 bytes.
/Users/glc/Library/Arduino15/packages/esp32/tools/esptool_py/2.6.1/esptool --chip esp32 --port /dev/cu.SLAB_USBtoUART --baud 115200 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size detect 0xe000 /Users/glc/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/partitions/boot_app0.bin 0x1000 /Users/glc/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/bin/bootloader_dio_80m.bin 0x10000 /var/folders/04/05cn9xk95tj9jvx4rv2_fvmr0000gn/T/arduino_build_835050/Encoder.ino.bin 0x8000 /var/folders/04/05cn9xk95tj9jvx4rv2_fvmr0000gn/T/arduino_build_835050/Encoder.ino.partitions.bin 
esptool.py v2.6
Serial port /dev/cu.SLAB_USBtoUART
Connecting........_____.
Chip is ESP32-PICO-D4 (revision 1)
Features: WiFi, BT, Dual Core, Embedded Flash, VRef calibration in efuse, Coding Scheme None
MAC: d8:a0:1d:61:11:9c
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 8192 bytes to 47...
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.0 seconds (effective 6421.9 kbit/s)...
Hash of data verified.
Compressed 15856 bytes to 10276...
Wrote 15856 bytes (10276 compressed) at 0x00001000 in 0.9 seconds (effective 139.2 kbit/s)...
Hash of data verified.
Compressed 231504 bytes to 114732...
Wrote 231504 bytes (114732 compressed) at 0x00010000 in 10.3 seconds (effective 180.4 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 128...
Wrote 3072 bytes (128 compressed) at 0x00008000 in 0.0 seconds (effective 1412.7 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...

madhephaestus, 2020-07-24:
does the latest tagged release work for you?

madhephaestus, 2020-07-24:
Also, what pins are you using?

madhephaestus, 2020-07-24:
Can you test using this bundled arduino/esp32 toolchain please?

https://github.com/WPIRoboticsEngineering/RobotInterfaceBoard/blob/master/InstallEclipse.md

madhephaestus, 2020-07-24:
I just did a test of the latest commit on hardware, the Sparkfun ESP32 Thing Plus, and the Romi encoders. Can not reproduce this issue.

myglc2, 2020-07-25:
I tried commit 4e8e228b19db2cb065f521358d078cdcc7134b3c
    #22 make the raw read private

I also doesn't work in what appears to be the same way.

I have not changed your example in any way. Pins are not connected to anything beyond those internal connections on the ESP32-PICO-KIT V4.1 dev board:

https://docs.espressif.com/projects/esp-idf/en/latest/esp32/hw-reference/esp32/get-started-pico-kit.html

Sorry, I don't do Windows.

I am actually using the ESP32 ESP-IDF tool chain.  I was trying your Arduino example to familiarize myself with ESP32 PCNT. So I don't need for you to make this work. Feel free to close/ignore this issue as you like. Best - George

closed 2020-08-13 by madhephaestus