madhephaestus/ESP32Encoder #30: Can only get 2 FullQuad encoders working.

stevecolt, 2020-08-13:
Hi, Great library, seems to work flawlessly on 2 channels, compiles but does nothing if I add a third. Is this a limitation on the library or is there something else I have to do for 3 or more encoders? I tried reading up on PCNT from link, and looking through the library, but can't see what resource difference is between Full or Half. I'm also struggling to see how to use channel 2. Is this magically done in the library ? 
Can I have 10 FullQuad encoders, or does type limit max?

ESP32Encoder xchan;    // pin21,22
ESP32Encoder ychan;   //18,19
ESP32Encoder zchan;   // 16,17

Then is setup
xchan.attachFullQuad(XApin,XBpin);
etc
It compiles without complaint if I use all 3 but only first 2 set work. this is true if I use xy, xz, yz 
Any suggestions?
Thanks
Steve

madhephaestus, 2020-08-13:
i have tested 4 encoders simultaneously, and it should support up to 10. Is it any set of pins not working? Will the non working encoder work if the others are not enabled? Is the non working one always the same one? Is it hooked up while testing it?

It could be current draw of the third device, it could be intermittent wiring. I have definitely used more than 2 successfully. 

Can you post a gist of repo of some project thats not working for you? Maybe a silly code mistake like reusing a pin in attachment, or attaching the same variable twice with different pins (copy paste error) perhaps?

thijstriemstra, 2020-08-13:
The readme mentions:

> 10 simultaneous quadrature encoders.

Does this then mean 40 single edge encoders? Or also 10?

madhephaestus, 2020-08-13:
There are only 10 PCNT channels, no matter what mode.

stevecolt, 2020-08-13:
Thanks for fast response.
Maybe I didn't make it clear, but any 2 out of the 3 channels did work, so yes the non working channel did work and vice versa, I dont think its a cut and paste problem, code is identical for each channel. At the moment I'm testing with a single rotary encoder and moving between channels. Were you using Half or Full quad? I'm trying to use full, but not sure from library what the difference is. I haven't yet tested to prove it is actually working in full quad mode, busy building a test box for that or use a scope and function generator.
Thanks
Steve

stevecolt, 2020-08-14:
Really sorry, my bad!! Looks like a tiepin error!!! 3 Channels working . Start testing prope.
Please close.
Apologies
Steve

closed 2020-08-14 by stevecolt