madhephaestus/ESP32Encoder #32: Full quad v single edge & voltage thresholds/pull down resistors

MoretonBayKiting, 2020-09-20:
Thanks for a great library. My apologies in advance if these questions should be elsewhere.

My implementation involves two IR photo-interrupters reading 2 windows in a flange mounted on the chuck of a drill.  The two windows provide sufficient accuracy as long as no counts are lost. An accurate neutral, for example, is important.   The size of the photo-interrupters necessitate windows of order 10mm long in a circumference of order 160mm.  When changing direction, the chuck can easily stop part way through a quadrature cycle - ie with a window over one or both detectors. Using the full quadrature implementation seems to work OK.  A single count per window would be easier. But will a single edge implementation be accurate?  Would it, for example, count on the way in as well as on the way out of the window if it stopped "inside".  For example, if the single edge counts rising edges, it might count one direction on the way in but nothing on the way out since there would be no rising edge?

I've had some success with these photo-interrupters with pull down resistors.  Do you recommend setting useInternalWeakPullResistors=DOWN; and not using any hardware pull downs?

Thanks in advance.

madhephaestus, 2020-09-22:
You want either full or half quadrature, because if it stops and reverses inside the window single edge mode wont read that correctly. Full is basically always better than not. 

As for the pullups, whatever works easiest for you, both external or internal would work for you.

closed 2020-09-22 by madhephaestus

MoretonBayKiting, 2020-09-22:
Thanks Kevin.