madhephaestus/ESP32Encoder #34: Example using interupts

targetarchitecture, 2020-11-12:
Hi Madhephaestus,
   I'm trying to build a event driven / interupt driven piece of code, I love your library and would like an example of how to get your code to fire on interupt , the only example uses the loop() method and that's not going to work for me.

Thanks

John

madhephaestus, 2020-11-12:
Well... there are no interrupts, the counting is done by a hardware peripheral. 

In fact, there are issues reading the encoder out of a timer, so i always suggest using a task like this:

https://github.com/WPIRoboticsEngineering/RBE1001Lib/blob/956e74a1389603b12f0e28eaaa349fc9ad89ec06/src/Motor.cpp#L77

and 

https://github.com/WPIRoboticsEngineering/RBE1001Lib/blob/956e74a1389603b12f0e28eaaa349fc9ad89ec06/src/Motor.cpp#L104

to get real-time PID as a FreeRTOS process. The Arduino stack already has FreeRTOS running, so adding tasks (Like threads) is super easy.

targetarchitecture, 2020-11-13:
No worries - just wrote my code a differant way

`#include <Arduino.h>
#include <ESP32Encoder.h>

ESP32Encoder encoder1;
ESP32Encoder encoder2;
volatile int32_t encoder1Count = 0;
volatile int32_t encoder2Count = 0;
TaskHandle_t EncodersTask;

void encoders_task(void *pvParameter);

void setup()
{
  Serial.begin(115200); //ESP32 USB Port

  // Enable the weak pull up resistors
  ESP32Encoder::useInternalWeakPullResistors = UP;

  // Attach pins for use as encoder pins
  encoder1.attachHalfQuad(36, 39);
  encoder1.clearCount();
  encoder1.setCount(0);

  // Attach pins for use as encoder pins
  encoder2.attachFullQuad(34, 35);
  encoder2.clearCount();
  encoder2.setCount(0);

  //start task to read an send rotary information every 500ms
  xTaskCreate(&encoders_task, "encoders_task", 2048, NULL, 0, NULL);
}

void encoders_task(void *pvParameter)
{
  std::string msg;
  int32_t newEncoder1Count;
  int32_t newEncoder2Count;

  while (1)
  {
    newEncoder1Count = encoder1.getCount();
    newEncoder2Count = encoder2.getCount();

    if (newEncoder1Count != encoder1Count)
    {
      msg = "D1,";

      if (newEncoder1Count > encoder1Count)
      {
        msg.append("+");
      }
      else
      {
        msg.append("-");
      }

      Serial.println("Encoder 2 = " + String(newEncoder1Count) + " direction = " + msg.c_str());

      encoder1Count = newEncoder1Count;
    }

    if (newEncoder2Count != encoder2Count)
    {
      msg = "D2,";

      if (newEncoder2Count > encoder2Count)
      {
        msg.append("+");
      }
      else
      {
        msg.append("-");
      }

      Serial.println("Encoder 2 = " + String(newEncoder2Count) + " direction = " + msg.c_str());

      encoder2Count = newEncoder2Count;
    }

    delay(50);
  }
}

void loop()
{
  delay(1000);
}
`

closed 2020-11-13 by targetarchitecture