madhephaestus/ESP32Encoder #35: counter from timer

jschoch, 2020-12-29:
I was reading the encoder count from a timer and it was working fine but it seems since some refactoring I have done I get some very big jumps between values.  I store the previous value and crash if it is few ticks more than the previous value.  The jumps are sometimes in the wrong direction.  I saw in another issue that you recommend using a task to read the value via getCount().  Since I need to make calculations on every encoder tick I wonder if you have any thoughts on how best to notify a task that the encoder has moved?  This doesn't seem possible with the current public methods and I'm guessing maybe pcnt isn't the right approach for me?

madhephaestus, 2020-12-29:
that is one feature that the PCNT module does not support. I think you can attach interrupts to the pins used for pcnt to get change events, the use pcnt for accurate counting.

Alternatively you can try using this interrupt driven encoder object with built in debouncing:

https://github.com/WPIRoboticsEngineering/RBE2001_ESP32_Solutions/blob/main/src/InterruptEncoder.cpp

jschoch, 2020-12-29:
Ok, thanks for the prompt reply!

closed 2020-12-29 by jschoch