madhephaestus/ESP32Encoder #36: Getting (xQueueGenericReceive)- assert failed!

sagreen83, 2021-01-14:
Still trying to track this down, but I seem to be getting random (xQueueGenericReceive)- assert failed! errors when I am spinning my rotary dial. Works for a lot of clicks then dies. Here is the error.

/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/queue.c:1446 (xQueueGenericReceive)- assert failed!
abort() was called at PC 0x4008880b on core 0

Backtrace: 0x4008c334:0x3ffbe030 0x4008c561:0x3ffbe050 0x4008880b:0x3ffbe070 0x400d45b9:0x3ffbe0b0 0x400d3761:0x3ffbe0d0 0x400d39ae:0x3ffbe0f0 0x40080ee1:0x3ffbe190 0x40080fdd:0x3ffbe1b0 0x40084ebd:0x3ffbe1d0 0x4000bfed:0x3ffbd090 0x40089c15:0x3ffbd0a0 0x40089231:0x3ffbd0c0 0x40089295:0x3ffbd0e0 0x4008afca:0x3ffbd100 0x4008b0ab:0x3ffbd140 0x40088ae5:0x3ffbd170

Rebooting...

madhephaestus, 2021-01-14:
can you please use the backtrace decoder: https://github.com/me-no-dev/EspExceptionDecoder and tell me precisely which line in this library generates the error?

sagreen83, 2021-02-04:
I think its safe to close this one. I've not been able to recreate nor has it happened since I opened this issue. My bad.

closed 2021-02-04 by madhephaestus