madhephaestus/ESP32Encoder #38: FR: Allow configuring filter value

positron96, 2021-03-11:
Hi. It would be very helpful to allow setting custom filter value (preferably in seconds or microseconds) to deal with noisy encoders. Right now it seems to be fixed at 250 ticks of APB_CLK which is 3.125 uS. In my case I could use a thousand times larger one.

madhephaestus, 2021-03-12:
That's a good idea, sure. Can you submit a PR adding that feature?

positron96, 2021-03-13:
Hi, thanks for the response. Upon closer inspection of documentation, I've realized that maximum working value is 1023, which is only 4 times larger than current 250. So it does not help with my encoder at all (to filter out anything smaller than 1 ms, I'd need a value of 80000). So, I can't use the PCNT peripheral.
I will provide a PR if time permits.

madhephaestus, 2021-03-14:
what sort of encoder has a 1ms noise? It might be worth looking into electrical de-bouncing then?

madhephaestus, 2021-03-14:
You might want to take a look at this lib i wrote to do custom debouncing and interrupt events on the ESP32:
https://github.com/WPIRoboticsEngineering/RBE2001_ESP32_Solutions/blob/main/src/InterruptEncoder.cpp

It will work the way you want but is less good than this lib at high-speed encoders. It might be just what you are looking for though...

commit e855b096 (2021-03-22): Merge pull request #39 from positron96/master

Add method to change filter value. Close #38
closed 2021-03-22 by madhephaestus