madhephaestus/ESP32Encoder #40: switch button

fotosettore, 2021-03-26:
hi !
finally the first library without bouncing problems !!!
will you also implement an internal routine that allows you to recognize the switch without having to build something external (digital state of any pin)?

madhephaestus, 2021-03-26:
digitalRead() works on pins used by the encoder library. If you want to send a PR ill add it :)

JimNickerson, 2021-03-27:
I have made use of digitalRead().
When using a KY-040 inexpensive mechanical rotary encoder I find it has very noisy edges.
I have increased the glitch filter using:
pcnt_set_filter_value(PCNT_UNIT_0, 1023); // ky-040 is mechanical encoder, use max glitch filter
This makes it better but still not without extra pulses from encoder
My desire is to use it with U8g2lib.h and the selectionlist example.
I replaced the weak function uint8_t u8x8_GetMenuEvent(u8x8_t *u8x8) with mine in the ino using encoder.
I have yet to get it all working, I am waiting for an optical encoder to try again

madhephaestus, 2021-03-28:
Ah, that thing again.... That device pops up in the issues again and again because it is so noisy. It needs electrical denouncing, it can not be done in software alone. Get a set of different size capicotors and start bug 100uf and work down to a few pf. Connect a cap on each encoder line to gnd. Watch the edge on a scope and  look for edge smoothing.

fotosettore, 2021-03-28:
.01 uF ... max .22
more no good ... it delays too much the response. 
But is solution not good for all situations ...

madhephaestus, 2021-03-28:
I added documentation to the README about KY-040 devices

commit 975a88c0 (2021-03-28): close #40

Add documentation about debouncing.
closed 2021-03-28 by madhephaestus