madhephaestus/ESP32Encoder #44: Using pcnt interrupt and more...

engineerarslan, 2021-07-14:
Dear Sir,

You said that you have used pcnt interrupt to prevent the 16 bit counter from overflowing. In my project I want to fire an interrupt when a certain value is reached for example 20000. In the ISR I want to reset the encoder count and give an output. Is it possible?

Secondly I wanted to read the step direction signal from a position controller using your library. For that I used it in "single" mode. It worked but I was only able to do it accurately for frequency<40 kHz. Higher than that and the pulses were missed. On the other hand I was able to accurately count and encoder at a frequency>300 kHz. What may be the reason?

madhephaestus, 2021-08-02:
It could be possible to set a custom overflow/underflow value, that would be a matter of changing from a constant to a variable and adding getter/setter to the variable. As for an event called from the ISR, that also is possible if you add a function pointer callback variable to the encoder object and add that event call to the ISR. This is a lot of work modding the library, but nothing you want is impossible. Start a PR and i can help advise you through the changes if you like?

larroy, 2022-02-01:
https://github.com/madhephaestus/ESP32E

larroy, 2022-02-03:
@engineerarslan but the PR hasn't been merged

engineerarslan, 2022-03-03:
Thanks Sir. Your library is great. Can you describe how to use this new feature?

madhephaestus, 2022-03-04:
The author of the new feature added an example, but it uses some specific LCD to work. @larroy would you mind making a new example that is only what is needed to demonstrate the feature? mayme measure time between ticks in the interrupt and print that to serial in a slow loop?

larroy, 2022-03-11:
Yes I can add another example, also add something to the readme. Just give me a few days since I'm busy with many other things.

larroy, 2022-03-11:
For the interest of OPs question, there's a callback in the constructor which you can pass, but be careful since it executes in ISR context, so writing to serial port or other complex stuff can crash the ESP32.