madhephaestus/ESP32Encoder #45: Encoder direction change = "Too many encoders, FAIL!"

KSSS1, 2021-08-02:
Hi,

I want to change encoder direction. I know it is possible to change this in software(SETUP) but i want to change on the fly in the void loop by attaching the pins reversed  "encoderX.attachHalfQuad(2, 3)" to  encoderX.attachHalfQuad(3, 2) When a button is pressed. This works **BUT**
when you do this a couple times in the loop then it gives on the serial monitor this **error** "Too many encoders, FAIL!" So when this error occurs there is no possibility to change the encoder directions anymore.

Is there a possibility to erase the ATTACHED encoders or to reset the encoder attached count? a funtion like "encoder_detach"? 

Thanks!

madhephaestus, 2021-08-02:
That is a use case i never thought of. Sure, go ahead and submit a PR for an encoder.detach() method and ill take a look :)

madhephaestus, 2021-12-19:
This feature is now availible in 0.7.0

commit 0ab7dcf8 (2021-12-19): Fix #45 by adding a detatch method
closed 2021-12-19 by madhephaestus