madhephaestus/ESP32Encoder #47: single pin encoder support

tinyboxxx, 2021-09-05:
Hi, Nice work!

I want use this in my speedo project, is it possible to attach a single pin encoder? I only need one direction.

I didn't find related ways to do it in document.

Thanks!

madhephaestus, 2021-09-07:
https://madhephaestus.github.io/ESP32Encoder/classESP32Encoder.html#ac565f0e078a89ab9ddcd8b6890365751

closed 2021-09-07 by madhephaestus

tinyboxxx, 2021-09-08:
thanks for your quick reply,

I did noticed that attachSingleEdge function, but one question: 
Since I only got one signal pin, but which pin should I put into bPinNumber ?

thanks again.

madhephaestus, 2021-09-08:
The same pin number :)

tinyboxxx, 2021-09-08:
huge Thanks!

tinyboxxx, 2021-09-15:
Hello, I got error when ESP32 booting, in serial.

```
E (719) pcnt: pcnt_unit_config(57): PCNT pluse input io error
E (720) pcnt: pcnt_unit_config(57): PCNT pluse input io error
E (720) pcnt: pcnt_unit_config(57): PCNT pluse input io error
E (725) pcnt: pcnt_unit_config(57): PCNT pluse input io error
```

ESP32 does keep running. Any idea what cause this error? Does some GPIO I can't use?
Thanks.

Code I use:

```
#include <ESP32Encoder.h>
ESP32Encoder encoder_speed;
ESP32Encoder encoder_rpm;

//in setup:

  ESP32Encoder::useInternalWeakPullResistors = UP;
  encoder_speed.attachSingleEdge(27, 27); // SPD
  encoder_rpm.attachSingleEdge(34, 34);   // RPM

//in loop:
        PulseCounter_SPD = (int32_t)encoder_speed.getCount();
        PulseCounter_RPM = (int32_t)encoder_rpm.getCount();
```