madhephaestus/ESP32Encoder #49: More than 8 encoders problem

matthgyver, 2021-12-17:
Hi,

I'm trying to have 9 encoders and the last one does'nt work.
There is my code :
```
#include <ESP32Encoder.h>     // https://github.com/madhephaestus/ESP32Encoder/
#include <BleGamepad.h>       // https://github.com/lemmingDev/ESP32-BLE-Gamepad

BleGamepad bleGamepad("SimRacing Wheel", "test", 100);

/*********************
 * JOYSTICK SETTINGS *
 *********************/
#define numOfButtons        40
#define numOfHatSwitches    0
#define enableX             false
#define enableY             false
#define enableZ             false
#define enableRZ            false
#define enableRX            false
#define enableRY            false
#define enableSlider1       false
#define enableSlider2       false
#define enableRudder        false
#define enableThrottle      false
#define enableAccelerator   false
#define enableBrake         false
#define enableSteering      false

#define MAXENC 9
uint8_t uppPin[MAXENC]     = {36,34,32,25,27,22,03,17,04};
uint8_t dwnPin[MAXENC]     = {39,35,33,26,14,01,21,16,00};
uint8_t encoderUpp[MAXENC] = {17,19,21,23,25,27,29,33,31};
uint8_t encoderDwn[MAXENC] = {18,20,22,24,26,28,30,34,32};
ESP32Encoder encoder[MAXENC];
unsigned long holdoff[MAXENC] = {0,0,0,0,0,0,0,0,0};
int32_t prevenccntr[MAXENC]   = {0,0,0,0,0,0,0,0,0};
bool prevprs[MAXENC]          = {0,0,0,0,0,0,0,0,0};
#define HOLDOFFTIME 150   // TO PREVENT MULTIPLE ROTATE "CLICKS" WITH CHEAP ENCODERS WHEN ONLY ONE CLICK IS INTENDED

////////////////////////////////////////////////////////////////////////////////////////

void sendKey(uint8_t key) {
    Serial.print("pulse\t");
    Serial.println(key);
    if(bleGamepad.isConnected()) {
      bleGamepad.press(key);
      delay(150);
      bleGamepad.release(key);
    }
}

void pressKey(uint8_t key) {
    Serial.print("hold\t");
    Serial.println(key);
    if(bleGamepad.isConnected()) {
      bleGamepad.press(key);
    }
}

void releaseKey(uint8_t key) {
    Serial.print("release\t");
    Serial.println(key);
    if(bleGamepad.isConnected()) {
      bleGamepad.release(key);
    }
}

////////////////////////////////////////////////////////////////////////////////////////

void setup() {
  Serial.begin(115200);

  for (uint8_t i=0; i<MAXENC; i++) {
    encoder[i].clearCount();
    encoder[i].attachSingleEdge(dwnPin[i], uppPin[i]);
  }
  bleGamepad.setControllerType(CONTROLLER_TYPE_GAMEPAD);  //CONTROLLER_TYPE_JOYSTICK, CONTROLLER_TYPE_GAMEPAD (DEFAULT), CONTROLLER_TYPE_MULTI_AXIS
  bleGamepad.begin(numOfButtons,numOfHatSwitches,enableX,enableY,enableZ,enableRZ,enableRX,enableRY,enableSlider1,enableSlider2,enableRudder,enableThrottle,enableAccelerator,enableBrake,enableSteering); 
  Serial.println("Booted!");
}

////////////////////////////////////////////////////////////////////////////////////////

void loop() {

  unsigned long now = millis();

  // -- ROTARY ENCODERS : ROTATION -- //

  for (uint8_t i=0; i<MAXENC; i++) {
    int32_t cntr = encoder[i].getCount();
    if (cntr!=prevenccntr[i]) {
      if (!holdoff[i]) {
        if (cntr>prevenccntr[i]) { sendKey(encoderUpp[i]); }
        if (cntr<prevenccntr[i]) { sendKey(encoderDwn[i]); }
        holdoff[i] = now;
        if (holdoff[i]==0) holdoff[i] = 1;  // SAFEGUARD WRAP AROUND OF millis() (WHICH IS TO 0) SINCE holdoff[i]==0 HAS A SPECIAL MEANING ABOVE
      }
      else if (now-holdoff[i] > HOLDOFFTIME) {
        prevenccntr[i] = encoder[i].getCount();
        holdoff[i] = 0;
      }
    }
  }

  delay(10);
 
}
```

The last defined encoder never work.
It's not a pin issue because if I change the order of my pin on ``uppPin`` and ``dwnPin``, the encoder with the last pin does'nt work.

I've make lot of test and check lot of things but I don't see where is the problem.

Have you an idea please ?

Thank's

madhephaestus, 2021-12-19:
this is a hardware peripheral and it only supports 8 encoders, no more. If you look in the log it prints a failure message when more are attached.

closed 2021-12-19 by madhephaestus

matthgyver, 2021-12-19:
Thank's for the precisions. The README need an update because it's written :
> support for up to 10 simultaneous quadrature encoders.

madhephaestus, 2021-12-19:
Ah, good catch, fixed that just now

engineerarslan, 2022-02-23:
In the title description still 10 encoders is mentioned