madhephaestus/ESP32Encoder #51: Compile error for ESP32-S2 due to variable-refactoring on esp32-arduino framework

systemofapwne, 2021-12-20:
Compiling this library on the arduino-esp32 framework for ESP32-S2 models fails, since a few variables have been renamed.

````.pio/libdeps/esp32-s2_otg/ESP32Encoder/src/ESP32Encoder.cpp: In function 'void pcnt_example_intr_handler(void*)':
.pio/libdeps/esp32-s2_otg/ESP32Encoder/src/ESP32Encoder.cpp:50:27: error: 'volatile union pcnt_un_status_reg_t' has no member named 'h_lim_lat'
    if(PCNT.status_unit[i].h_lim_lat){
                           ^~~~~~~~~
.pio/libdeps/esp32-s2_otg/ESP32Encoder/src/ESP32Encoder.cpp:53:27: error: 'volatile union pcnt_un_status_reg_t' has no member named 'l_lim_lat'
    if(PCNT.status_unit[i].l_lim_lat){
                           ^~~~~~~~~
````

ESP32: https://github.com/espressif/arduino-esp32/blob/d6934a528941f1ee545917c0a61e98d1c23fcc83/tools/sdk/esp32/include/soc/esp32/include/soc/pcnt_struct.h#L128

ESP32-S2: https://github.com/espressif/arduino-esp32/blob/d6934a528941f1ee545917c0a61e98d1c23fcc83/tools/sdk/esp32s2/include/soc/esp32s2/include/soc/pcnt_struct.h#L251

My current workaround is to use a preprocessor directive to globally rename the new variable name to the old one.
It would be however more elegant to directly fix this within this library by checking for the ESP32-S2 model and using the correct variable names when running on the ESP32-S2 platform.

madhephaestus, 2021-12-22:
using #ifdef's like this:
https://github.com/madhephaestus/ESP32AnalogRead/blob/master/ESP32AnalogRead.cpp#L132
is how to make a single code base cross-compile for different ESP32 versions.

madhephaestus, 2021-12-22:
If you send a PR with that #ifdef gaurds then i can merge that in :)

systemofapwne, 2021-12-27:
I added the changes to my previous PR regarding issues on the ESP32-S2 platform and renamed the PR to "ESP32-S2 platform support".

Feel free to pull.

brianmichalk1, 2022-04-05:
I think I am having a similar problem:
```
.pio/libdeps/esp32-s2-saola-1a/ESP32Encoder/src/ESP32Encoder.cpp: In function 'void esp32encoder_pcnt_intr_handler(void*)':
.pio/libdeps/esp32-s2-saola-1a/ESP32Encoder/src/ESP32Encoder.cpp:68:65: error: 'volatile union pcnt_un_status_reg_t' has no member named 'thres0_lat'
    } else if(esp32enc->always_interrupt && (PCNT.status_unit[i].thres0_lat || PCNT.status_unit[i].thres1_lat)) {
```
platformio.ini:
```

[env:esp32-s2-saola-1a]
platform = espressif32
board = esp32-s2-saola-1
framework = arduino
lib_deps = 
	mcxiaoke/ESPDateTime
	madhephaestus/ESP32Encoder
	hieromon/AutoConnect
	adafruit/Adafruit SleepyDog Library
	EEPROM
build_flags = 
	${common_env_data.build_flags}
```
```
Dependency Graph
|-- <ESPDateTime> 1.0.4
|-- <ESP32Encoder> 0.9.0
|-- <AutoConnect> 1.3.4
|   |-- <PageBuilder> 1.5.3
|   |   |-- <FS> 2.0.0
|   |   |-- <LittleFS> 2.0.0
|   |   |   |-- <FS> 2.0.0
|   |   |-- <SPIFFS> 2.0.0
|   |   |   |-- <FS> 2.0.0
|   |   |-- <WebServer> 2.0.0
|   |   |   |-- <WiFi> 2.0.0
|   |   |   |-- <FS> 2.0.0
|   |   |-- <WiFi> 2.0.0
|   |-- <ArduinoJson> 6.19.4
|   |-- <SD> 2.0.0
|   |   |-- <FS> 2.0.0
|   |   |-- <SPI> 2.0.0
|   |-- <SPI> 2.0.0
|   |-- <WiFi> 2.0.0
|   |-- <FS> 2.0.0
|   |-- <LittleFS> 2.0.0
|   |   |-- <FS> 2.0.0
|   |-- <SPIFFS> 2.0.0
|   |   |-- <FS> 2.0.0
|   |-- <WebServer> 2.0.0
|   |   |-- <WiFi> 2.0.0
|   |   |-- <FS> 2.0.0
|   |-- <EEPROM> 2.0.0
|   |-- <Preferences> 2.0.0
|   |-- <Update> 2.0.0
|   |-- <Ticker> 2.0.0
|   |-- <HTTPClient> 2.0.0
|   |   |-- <WiFi> 2.0.0
|   |   |-- <WiFiClientSecure> 2.0.0
|   |   |   |-- <WiFi> 2.0.0
|   |-- <HTTPUpdate> 2.0.0
|   |   |-- <HTTPClient> 2.0.0
|   |   |   |-- <WiFi> 2.0.0
|   |   |   |-- <WiFiClientSecure> 2.0.0
|   |   |   |   |-- <WiFi> 2.0.0
|   |   |-- <Update> 2.0.0
|   |   |-- <WiFi> 2.0.0
|   |-- <DNSServer> 2.0.0
|   |   |-- <WiFi> 2.0.0
|-- <Adafruit SleepyDog Library> 1.6.0
|-- <EEPROM> 2.0.0
|-- <WebServer> 2.0.0
|   |-- <WiFi> 2.0.0
|   |-- <FS> 2.0.0
|-- <WiFi> 2.0.0
```

Sp-Link, 2022-04-17:
I am having a same problem, which was solved after modifying the code. The chip used is esp-s3.How should I send the code?

Sp-Link, 2022-04-17:
I am having a same problem, which was solved after modifying the code. The chip used is esp-s3.It may be wrong, but the example running.How should I send the code?

systemofapwne, 2022-04-17:
> I am having a same problem, which was solved after modifying the code. The chip used is esp-s3.It may be wrong, but the example running.How should I send the code?

Just via a standard Pull Request (PR): Fork the repo, clone it, change the code, push the changes and apply for a PR.

madhephaestus, 2022-08-05:
this should be resolved now

closed 2022-08-05 by madhephaestus