madhephaestus/ESP32Encoder #62: Reading Step/Dir inputs

MadTooler, 2022-06-08:
I have been using the library to read my full quadrature encoders with very good results.  

My current project is a custom closed loop stepper motion control.  The esp32 will receive step/dir inputs for two motors, output to the drivers, and monitor the two related encoders for closed loop control.  There are other tasks and features the system will be performing that are above what a stock closed loop controller would provide.

Since I am only reading two encoders, I thought maybe a tweak to the library for a new encoder mode would make it possible to also read step and direction inputs?

Thanks.

madhephaestus, 2022-06-11:
Sure, not a bad idea at all. There are a few different modes in the lib, it would be fairly straight forward for you to add a new mode and submit a Pull Request.

MadTooler, 2022-06-11:
I have tried the different modes and had some success with single edge.  I also tried tweaking the library without much luck.  A lot of your coding is outside of my current limited understanding of assumed low level higher efficiency and/or esp specific.  I spend most of my time with higher level code without as good of a grasp of the nuts and bolts that make it work so well.

Side note, looking around, I have not found a description of how the different encoder modes count or any limitations with specific GPIO pins.  Do you have that posted somewhere?

Thanks again!

MadTooler, 2022-06-14:
It looks like using the mode attachSingleEdge(aPin, bPin) is doing the trick.  Steps to aPin and Direction to bPin.

Looking over the library and with testing, I think single edge only counts the rising edge of aPin (steps) and increments the count per state of bPin.  Is that correct, or is there some extra increments bouncing back and forth that I am missing?

With my bench tests and debounce timing settings in the library, high speed stepping appears limited.  My application is pretty slow, but others may not be able to make use.  I also have had some erratic loss of observed steps, but that probably is loose jumper wires/breadboard on my test or the lacking MCU/stepper library I was initially using to simulate steps.

madhephaestus, 2022-06-20:
Sounds like this is resolved then :)

closed 2022-06-20 by madhephaestus