madhephaestus/ESP32Encoder #79: Compilation fails in Arduino IDE 2.0 for esp32c3

jackjansen, 2022-10-07:
I'm trying to build for an esp32c3 for the first time, and I get the error

```
Documents/Arduino/libraries/ESP32Encoder/src/ESP32Encoder.cpp:9:10: fatal error: soc/pcnt_struct.h: No such file or directory
| #include <soc/pcnt_struct.h>
```

I'm not sure whether this is a serious problem and esp32encoder cannot be compatible with the esp32c3 because it is missing the pulse counter hardware or some other essential thing, or that it is a one-line fix, simply including a different header or some such.

jackjansen, 2022-10-07:
Answering my own question: it seems the esp32c3 doesn't have the pulse counter hardware, according to https://docs.espressif.com/projects/esp-idf/en/v4.3/esp32c3/hw-reference/chip-series-comparison.html

So I guess I'll just have to use my legacy esp8266 pulse counting code on that chip.

It might be worthwhile to mention this in the ESP32Encoder documentation.

madhephaestus, 2022-10-08:
you can use the interrupt encoder code, that doesnt rely on the pulse counter

closed 2022-10-08 by madhephaestus