madhephaestus/ESP32Encoder #84: Not compiling for ESP32-S2

adaand00, 2022-11-24:
Hi, I am having trouble compiling for a ESP32-S2 using platformIO. I get the following error:
```
.pio/libdeps/Esp32-s2/ESP32Encoder/src/ESP32Encoder.cpp: In member function 'void ESP32Encoder::attach(int, int, encType)':
.pio/libdeps/Esp32-s2/ESP32Encoder/src/ESP32Encoder.cpp:210:19: error: 'esp_ipc_call_blocking' was not declared in this scope
```

My guess is that this is because the S2 only has a single core and therefore `esp_ipc_call_blocking()` is meaningless. It works using release 0.9.1 (using lib_deps = madhephaestus/ESP32Encoder@0.9 in platformio.ini). `esp_ipc.h` seems to use the macro  `CONFIG_FREERTOS_UNICORE`, perhaps that would work here too.

cdromcdrom, 2023-08-15:
Same problem here，have you solved it？ @adaand00 
Error code here：
`c:\Users\Administrator\Documents\Arduino\libraries\ESP32Encoder\src\ESP32Encoder.cpp: In member function 'void ESP32Encoder::attach(int, int, encType)':
c:\Users\Administrator\Documents\Arduino\libraries\ESP32Encoder\src\ESP32Encoder.cpp:210:19: error: 'esp_ipc_call_blocking' was not declared in this scope
    esp_err_t er = esp_ipc_call_blocking(isrServiceCpuCore, ipc_install_isr_on_core, &ipc_ret_code);
                   ^~~~~~~~~~~~~~~~~~~~~

exit status 1

Compilation error: exit status 1`

I‘m using release 0.10.1, the board is also ESP32-S2.
Changing the version is useless for me.

Tymothy, 2024-02-29:
For anyone looking in the future, ESP32-S2 is single core, and the 'esp_ipc_call_blocking' line would prevent my code from building.  I ended up deleting pretty much all the code in that if statement in ESP32Encoder.cpp and it seems to work fine for me.  It looks like hippo5329 has a pull request to account for this in a better way than I handled it.

https://github.com/madhephaestus/ESP32Encoder/pull/102

closed 2024-02-29 by madhephaestus