madhephaestus/ESP32Encoder #85: Compile error 0.10.0 and 0.10.1

Michiboi29, 2022-11-27:
The Version 0.10.0 and 0.10.1 seems to require "soc/soc_caps.h" which I don't believe it should.

`|-- ESP32Encoder @ 0.10.1
Building in debug mode
Compiling .pio\build\esp32dev\lib847\ESP32Encoder\ESP32Encoder.cpp.o
Compiling .pio\build\esp32dev\FrameworkArduino\IPv6Address.cpp.o
Compiling .pio\build\esp32dev\FrameworkArduino\MD5Builder.cpp.o
.pio\libdeps\esp32dev\ESP32Encoder\src\ESP32Encoder.cpp:9:26: fatal error: soc/soc_caps.h: No such file or directory
compilation terminated.
Compiling .pio\build\esp32dev\FrameworkArduino\Print.cpp.o
Compiling .pio\build\esp32dev\FrameworkArduino\Stream.cpp.o
Compiling .pio\build\esp32dev\FrameworkArduino\StreamString.cpp.o
*** [.pio\build\esp32dev\lib847\ESP32Encoder\ESP32Encoder.cpp.o] Error 1`

MichMich, 2022-12-03:
Same issue here. Reverting back to `0.9.1` works.

CafeNoir64, 2022-12-14:
me three, 0.9 also works for me

byetool, 2023-01-09:
the same problem to me  ,how to do ?

byetool, 2023-01-09:
Compiling .pio\build\release\lib4b6\SPI\SPI.cpp.o
Compiling .pio\build\release\libb76\TMCStepper\source\CHOPCONF.cpp.o
Compiling .pio\build\release\libb76\TMCStepper\source\COOLCONF.cpp.o
.pio\libdeps\release\ESP32Encoder\src\ESP32Encoder.cpp:9:26: fatal error: soc/soc_caps.h: No such file or directory
compilation terminated.

jackjansen, 2023-04-07:
I'm having the same problem, but strangely enough only when I build with `Arduino-cli`. When I build with `platformio` things seem to work fine for me... And indeed I have the `soc_caps.h` file in my platform:

```
jack@sap:.platformio% find . -print | grep soc_caps.h
./packages/framework-arduinoespressif32/tools/sdk/esp32c3/include/soc/esp32c3/include/soc/soc_caps.h
./packages/framework-arduinoespressif32/tools/sdk/esp32/include/soc/esp32/include/soc/soc_caps.h
./packages/framework-arduinoespressif32/tools/sdk/esp32s3/include/soc/esp32s3/include/soc/soc_caps.h
./packages/framework-arduinoespressif32/tools/sdk/esp32s2/include/soc/esp32s2/include/soc/soc_caps.h
jack@sap:.platformio%
```

jackjansen, 2023-04-12:
After a lot of comparing my ESP-IDF trees I get the impression that `soc/soc_caps.h` was introduced in IDF 4.0, so if you're using IDF 3.3 or earlier that file doesn't exist. 

I'm experimenting with something like

```
#include <esp_idf_version.h>
#if ESP_IDF_VERSION >= ESP_IDF_VERSION_VAL(4, 0, 0) 
#include "soc/soc_caps.h"
// Use the SOC capability defines to determine what this esp32 variant supports
#else
// Make some wild guess
#endif
```

But the "wild guess" is a bit of a problem: various of the newer esp32 variants (such as the esp32c3 don't support the PCNT hardware, and I don't think there's a way to find out if you don't have soc_caps.h.