madhephaestus/ESP32Encoder #88: Encoder not working in IO2, IO4.

ztjona, 2023-05-15:
I am using 4 full quad. encoders simultaneously, 3 encoders work fine, but one of them do not.

I am using a Wemos D1 R32 (the one that has the shape of the Arduino Uno but with an ESP32).

The encoder that does not work is in the pins IO2 - IO4 (in the UNO shape A0 - A1). 
The encoders that work are in the pins: 
35-34, 36-39, 26-18 (A2-A3,  A4-A5, 13-2).

What happens in these pins that the encoder does not work? Can I fix it somehow?

PD: I can not use other pins for those will be used by the Adafruit Motor Shield V1.

madhephaestus, 2023-05-15:
have you verified with a scope that the pins are working as expected? 

Are you using pull-ups/downs?

Check the datasheet for other peripherals that may be using those pins. There is no obvious reason it should not work.

madhephaestus, 2023-05-24:
love a solution like that!