madhephaestus/ESP32Encoder #89: Compile fails with soc.h not found

nicechocolate, 2023-05-22:
Sorry to interrupt you, but I had this when trying to update an old project (yes, it does have a coding bug that I didn't find two years ago...)

I updated ESP32Encoder to the latest version 0.10.1 and I get "soc.h not found" when trying to compile with Arduino IDE 2.1.0). I have downgraded to version 0.6.0 and it compiles fine. 

I shouldn't need to recompile this project again and anyway, I don't mind changing back when I need to, having made a note in my program's comments, but I didn't see anything like this mentioned in any other issues or your read.me, (though it could well be there, my concentration span is not what it was before I had Covid.)

MartinNohr, 2023-10-06:
I'm having the same problem. soc.h not found.

madhephaestus, 2023-10-11:
what chip are you compiling for?

I am working now to update my libraries to support the new toolchains and finding a lot of differences in what the toolchain provides.

nicechocolate, 2023-10-12:
Kevin, thank you for replying.

I'm using ESP32 DOIT Devkit 1 which has a WROOM-32. I was using the latest (at the time) version of the Arduino IDE, though it has updated since I got the error message.

I haven't tried any other processors, and anything could have changed since I posted the query, because I have had no need to compile that project again. I occasionally use ESP8266 D1 Mini, but now I expect to use the smaller ESP32-S3 for those smaller projects.

Just in case nobody else says it, THANK YOU. I really appreciate the efforts of anyone who, as you have, makes their libraries public. I have all on just remembering how to program, now I'm a bit older and 40 years of computer games has turned my brain to jelly. Covid didn't do anything for my concentration, either.

Neil
PS Other libraries, in case that's useful:
#include <SPI.h>
#include <TFT_eSPI.h> 
#include <WiFi.h>
#include "TimeLib.h"
#include <Preferences.h>
#include <FastLED.h>
#include <TM1637Display.h>
#include <Button.h>  // Button by Michael Adams. Best library I've found for this.
#include "free_fonts.h"
#include <ESP32Encoder.h>

I also replied to the email notification of this comment, goodness knows where that went!

f2knpw, 2023-11-07:
Hi,
Thanks for this lib.

But I do have the same issue "soc.h"
I am compiling for a Wemos lolin32 lite board with Arduino 1.8.19

Thanks for your help
JP

codefaux, 2024-03-22:
Same problem. Anyone figure a workaround? Anyone know of an Encoder library compatible with ESP32 which Just Works(tm)?

remcoder, 2024-08-24:
Had the same issue. After reading #85 was able to fix this compile error by explicitly building against Expressif 32 v4 or higher.

If you're using PlatformIO update the platform in platform.ini:
`platform = espressif32@4.0.0`

HTH!