madhephaestus/ESP32Encoder #91: Compiling error

Blaubart, 2023-08-04:
I updated my Arduino IDE and now I'm no longer able to compile.

animevietsub, 2023-08-04:
Can you give me all of the error log? I've just compiled it with no problem.

Blaubart, 2023-08-04:
Hi,

thanks for your fast replay!!

How can I generate more information with Arduino IDE??

Regards
Dirk

animevietsub, 2023-08-04:
Enable show verbose output during `compile`:
[image]
And copy all of the Output:
[image]

Blaubart, 2023-08-04:
Hi,

hope i helps:

Arduino: 1.8.13 (Mac OS X), Board: "DOIT ESP32 DEVKIT V1, 80MHz, 115200, None"

/Applications/Arduino.app/Contents/Java/arduino-builder -dump-prefs -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/dirk/Library/Arduino15/packages -hardware /Users/dirk/Documents/Arduino/hardware -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/dirk/Library/Arduino15/packages -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/dirk/Documents/Arduino/libraries -fqbn=esp32:esp32:esp32doit-devkit-v1:FlashFreq=80,UploadSpeed=115200,DebugLevel=none -vid-pid=0000_0000 -ide-version=10813 -build-path /var/folders/6t/qjgfw2413f7ddnjzk_08btxw0000gp/T/arduino_build_173134 -warnings=none -build-cache /var/folders/6t/qjgfw2413f7ddnjzk_08btxw0000gp/T/arduino_cache_278887 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.esptool_py.path=/Users/dirk/Library/Arduino15/packages/esp32/tools/esptool_py/2.6.1 -prefs=runtime.tools.esptool_py-2.6.1.path=/Users/dirk/Library/Arduino15/packages/esp32/tools/esptool_py/2.6.1 -prefs=runtime.tools.mkspiffs.path=/Users/dirk/Library/Arduino15/packages/esp32/tools/mkspiffs/0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=/Users/dirk/Library/Arduino15/packages/esp32/tools/mkspiffs/0.2.3 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=/Users/dirk/Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-80-g6c4433a-5.2.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc-1.22.0-80-g6c4433a-5.2.0.path=/Users/dirk/Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-80-g6c4433a-5.2.0 -verbose /Users/dirk/Documents/GitHub/FreeVarioGauge/src/FreeVarioGauge/FreeVarioGauge.ino
/Applications/Arduino.app/Contents/Java/arduino-builder -compile -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/dirk/Library/Arduino15/packages -hardware /Users/dirk/Documents/Arduino/hardware -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/dirk/Library/Arduino15/packages -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/dirk/Documents/Arduino/libraries -fqbn=esp32:esp32:esp32doit-devkit-v1:FlashFreq=80,UploadSpeed=115200,DebugLevel=none -vid-pid=0000_0000 -ide-version=10813 -build-path /var/folders/6t/qjgfw2413f7ddnjzk_08btxw0000gp/T/arduino_build_173134 -warnings=none -build-cache /var/folders/6t/qjgfw2413f7ddnjzk_08btxw0000gp/T/arduino_cache_278887 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.esptool_py.path=/Users/dirk/Library/Arduino15/packages/esp32/tools/esptool_py/2.6.1 -prefs=runtime.tools.esptool_py-2.6.1.path=/Users/dirk/Library/Arduino15/packages/esp32/tools/esptool_py/2.6.1 -prefs=runtime.tools.mkspiffs.path=/Users/dirk/Library/Arduino15/packages/esp32/tools/mkspiffs/0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=/Users/dirk/Library/Arduino15/packages/esp32/tools/mkspiffs/0.2.3 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=/Users/dirk/Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-80-g6c4433a-5.2.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc-1.22.0-80-g6c4433a-5.2.0.path=/Users/dirk/Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-80-g6c4433a-5.2.0 -verbose /Users/dirk/Documents/GitHub/FreeVarioGauge/src/FreeVarioGauge/FreeVarioGauge.ino
Using board 'esp32doit-devkit-v1' from platform in folder: /Users/dirk/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4
Using core 'esp32' from platform in folder: /Users/dirk/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4
Detecting libraries used...
/Users/dirk/Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-80-g6c4433a-5.2.0/bin/xtensa-esp32-elf-g++ -DESP_PLATFORM "-DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"" -DHAVE_CONFIG_H -DGCC_NOT_5_2_0=0 -DWITH_POSIX -I/Users/dirk/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/config -I/Users/dirk/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/app_trace -I/Users/dirk/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/app_update -I/Users/dirk/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/asio -I/Users/dirk/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/bootloader_support -I/Users/dirk/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/bt -I/Users/dirk/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/coap -I/Users/dirk/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/console -I/Users/dirk/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/driver -I/Users/dirk/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/esp-tls -I/Users/dirk/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/esp32 -I/Users/dirk/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/esp_adc_cal -I/Users/dirk/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/esp_event -I/Users/dirk/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/esp_http_client -I/Users/dirk/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/esp_http_server -I/Users/dirk/Library/Arduino15/packages/esp32/hardware/esp32/1.0. [... cut ...] sers/dirk/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/xtensa-debug-module -I/Users/dirk/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/esp-face -I/Users/dirk/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/esp32-camera -I/Users/dirk/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/esp-face -I/Users/dirk/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/fb_gfx -std=gnu++11 -Os

Blaubart, 2023-08-04:
Hi,

I updated all libraries and now I get a different error:
Arduino: 1.8.13 (Mac OS X), Board: "DOIT ESP32 DEVKIT V1, 80MHz, 115200, None, Disabled"

/Applications/Arduino.app/Contents/Java/arduino-builder -dump-prefs -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/dirk/Library/Arduino15/packages -hardware /Users/dirk/Documents/Arduino/hardware -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/dirk/Library/Arduino15/packages -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/dirk/Documents/Arduino/libraries -fqbn=esp32:esp32:esp32doit-devkit-v1:FlashFreq=80,UploadSpeed=115200,DebugLevel=none,EraseFlash=none -vid-pid=0000_0000 -ide-version=10813 -build-path /var/folders/6t/qjgfw2413f7ddnjzk_08btxw0000gp/T/arduino_build_710579 -warnings=none -build-cache /var/folders/6t/qjgfw2413f7ddnjzk_08btxw0000gp/T/arduino_cache_690012 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=/Users/dirk/Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc-esp-2021r2-patch5-8.4.0.path=/Users/dirk/Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0 -prefs=runtime.tools.riscv32-esp-elf-gcc.path=/Users/dirk/Library/Arduino15/packages/esp32/tools/riscv32-esp-elf-gcc/esp-2021r2-patch5-8.4.0 -prefs=runtime.tools.riscv32-esp-elf-gcc-esp-2021r2-patch5-8.4.0.path=/Users/dirk/Library/Arduino15/packages/esp32/tools/riscv32-esp-elf-gcc/esp-2021r2-patch5-8.4.0 -prefs=runtime.tools.mklittlefs.path=/Users/dirk/Library/Arduino15/packages/esp32/tools/mklittlefs/3.0.0-gnu12-dc7f933 -prefs=runtime.tools.mklittlefs-3.0.0-gnu12-dc7f933.path=/Users/dirk/Library/Arduino15/packages/esp32/tools/mklittlefs/3.0.0-gnu12-dc7f933 -prefs=runtime.tools.dfu-util.path=/Users/dirk/Library/Arduino15/packages/arduino/tools/dfu-util/0.11.0-arduino5 -prefs=runtime.tools.dfu-util-0.11.0-arduino5.path=/Users/dirk/Library/Arduino15/packages/arduino/tools/dfu-util/0.11.0-arduino5 -prefs=runtime.tools.xtensa-esp32s3-elf-gcc.path=/Users/dirk/Library/Arduino15/packages/esp32/tools/xtensa-esp32s3-elf-gcc/esp-2021r2-patch5-8.4.0 -prefs=runtime.tools.xtensa-esp32s3-elf-gcc-esp-2021r2-patch5-8.4.0.path=/Users/dirk/Library/Arduino15/packages/esp32/tools/xtensa-esp32s3-elf-gcc/esp-2021r2-patch5-8.4.0 -prefs=runtime.tools.openocd-esp32.path=/Users/dirk/Library/Arduino15/packages/esp32/tools/openocd-esp32/v0.11.0-esp32-20221026 -prefs=runtime.tools.openocd-esp32-v0.11.0-esp32-20221026.path=/Users/dirk/Library/Arduino15/packages/esp32/tools/openocd-esp32/v0.11.0-esp32-20221026 -prefs=runtime.tools.riscv32-esp-elf-gdb.path=/Users/dirk/Library/Arduino15/packages/esp32/tools/riscv32-esp-elf-gdb/11.2_20220823 -prefs=runtime.tools.riscv32-esp-elf-gdb-11.2_20220823.path=/Users/dirk/Library/Arduino15/packages/esp32/tools/riscv32-esp-elf-gdb/11.2_20220823 -prefs=runtime.tools.xtensa-esp32s2-elf-gcc.path=/Users/dirk/Library/Arduino15/packages/esp32/tools/xtensa-esp32s2-elf-gcc/esp-2021r2-patch5-8.4.0 -prefs=runtime.tools.xtensa-esp32s2-elf-gcc-esp-2021r2-patch5-8.4.0.path=/Users/dirk/Library/Arduino15/packages/esp32/tools/xtensa-esp32s2-elf-gcc/esp-2021r2-patch5-8.4.0 -prefs=runtime.tools.xtensa-esp-elf-gdb.path=/Users/dirk/Library/Arduino15/packages/esp32/tools/xtensa-esp-elf-gdb/11.2_20220823 -prefs=runtime.tools.xtensa-esp-elf-gdb-11.2_20220823.path=/Users/dirk/Library/Arduino15/packages/esp32/tools/xtensa-esp-elf-gdb/11.2_20220823 -prefs=runtime.tools.esptool_py.path=/Users/dirk/Library/Arduino15/packages/esp32/tools/esptool_py/4.5.1 -prefs=runtime.tools.esptool_py-4.5.1.path=/Users/dirk/Library/Arduino15/packages/esp32/tools/esptool_py/4.5.1 -prefs=runtime.tools.mkspiffs.path=/Users/dirk/Library/Arduino15/packages/esp32/tools/mkspiffs/0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=/Users/dirk/Library/Arduino15/packages/esp32/tools/mkspiffs/0.2.3 -verbose /Users/dirk/Documents/GitHub/FreeVarioGauge/src/FreeVarioGauge/FreeVarioGauge.ino
/Applications/Arduino.app/Contents/Java/arduino-builder -compile -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/dirk/Library/Arduino15/packages -hardware /Users/dirk/Documents/Arduino/hardware -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/dirk/Library/Arduino15/packages -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/dirk/Documents/Arduino/libraries -fqbn=esp32:esp32:esp32doit-devkit-v1:FlashFreq=80,UploadSpeed=115200,DebugLevel=none,EraseFlash=none -vid-pid=0000_0000 -ide-version=10813 -build-path /var/folders/6t/qjgfw2413f7ddnjzk_08btxw0000gp/T/arduino_build_710579 -warnings=none -build-cache /var/folders/6t/qjgfw2413f7ddnjzk_08btxw0000gp/T/arduino_cache_690012 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=/Users/dirk/Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc-esp-2021r2-patch5-8.4.0.path=/Users/dirk/Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0 -prefs=runtime.tools.riscv32-esp-elf-gcc.path=/Users/dirk/Library/Arduino15/packages/esp32/tools/riscv32-esp-elf-gcc/esp-2021r2-patch5-8.4.0 -prefs=runtime.tools.r [... cut ...] 32/include/mbedtls/mbedtls/include -I/Users/dirk/Library/Arduino15/packages/esp32/hardware/esp32/2.0.11/tools/sdk/esp32/include/mbedtls/esp_crt_bundle/include -I/Users/dirk/Library/Arduino15/packages/esp32/hardware/esp32/2.0.11/tools/sdk/esp32/include/app_update/include -I/Users/dirk/Library/Arduino15/packages/esp32/hardware/esp32/2.0.11/tools/sdk/esp32/include/spi_flash/include -I/Users/dirk/Library/Arduino15/packages/esp32/hardware/esp32/2.0.11/tools/sdk/esp32/include/bootloader_support/includ

Blaubart, 2023-08-04:
Hi,

 

I updated all libraries and now I get a different error:

 

Arduino: 1.8.13 (Mac OS X), Board: "DOIT ESP32 DEVKIT V1, 80MHz, 115200, None, Disabled"

/Applications/Arduino.app/Contents/Java/arduino-builder -dump-prefs -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/dirk/Library/Arduino15/packages -hardware /Users/dirk/Documents/Arduino/hardware -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/dirk/Library/Arduino15/packages -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/dirk/Documents/Arduino/libraries -fqbn=esp32:esp32:esp32doit-devkit-v1:FlashFreq=80,UploadSpeed=115200,DebugLevel=none,EraseFlash=none -vid-pid=0000_0000 -ide-version=10813 -build-path /var/folders/6t/qjgfw2413f7ddnjzk_08btxw0000gp/T/arduino_build_710579 -warnings=none -build-cache /var/folders/6t/qjgfw2413f7ddnjzk_08btxw0000gp/T/arduino_cache_690012 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=/Users/dirk/Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc-esp-2021r2-patch5-8.4.0.path=/Users/dirk/Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0 -prefs=runtime.tools.riscv32-esp-elf-gcc.path=/Users/dirk/Library/Arduino15/packages/esp32/tools/riscv32-esp-elf-gcc/esp-2021r2-patch5-8.4.0 -prefs=runtime.tools.riscv32-esp-elf-gcc-esp-2021r2-patch5-8.4.0.path=/Users/dirk/Library/Arduino15/packages/esp32/tools/riscv32-esp-elf-gcc/esp-2021r2-patch5-8.4.0 -prefs=runtime.tools.mklittlefs.path=/Users/dirk/Library/Arduino15/packages/esp32/tools/mklittlefs/3.0.0-gnu12-dc7f933 -prefs=runtime.tools.mklittlefs-3.0.0-gnu12-dc7f933.path=/Users/dirk/Library/Arduino15/packages/esp32/tools/mklittlefs/3.0.0-gnu12-dc7f933 -prefs=runtime.tools.dfu-util.path=/Users/dirk/Library/Arduino15/packages/arduino/tools/dfu-util/0.11.0-arduino5 -prefs=runtime.tools.dfu-util-0.11.0-arduino5.path=/Users/dirk/Library/Arduino15/packages/arduino/tools/dfu-util/0.11.0-arduino5 -prefs=runtime.tools.xtensa-esp32s3-elf-gcc.path=/Users/dirk/Library/Arduino15/packages/esp32/tools/xtensa-esp32s3-elf-gcc/esp-2021r2-patch5-8.4.0 -prefs=runtime.tools.xtensa-esp32s3-elf-gcc-esp-2021r2-patch5-8.4.0.path=/Users/dirk/Library/Arduino15/packages/esp32/tools/xtensa-esp32s3-elf-gcc/esp-2021r2-patch5-8.4.0 -prefs=runtime.tools.openocd-esp32.path=/Users/dirk/Library/Arduino15/packages/esp32/tools/openocd-esp32/v0.11.0-esp32-20221026 -prefs=runtime.tools.openocd-esp32-v0.11.0-esp32-20221026.path=/Users/dirk/Library/Arduino15/packages/esp32/tools/openocd-esp32/v0.11.0-esp32-20221026 -prefs=runtime.tools.riscv32-esp-elf-gdb.path=/Users/dirk/Library/Arduino15/packages/esp32/tools/riscv32-esp-elf-gdb/11.2_20220823 -prefs=runtime.tools.riscv32-esp-elf-gdb-11.2_20220823.path=/Users/dirk/Library/Arduino15/packages/esp32/tools/riscv32-esp-elf-gdb/11.2_20220823 -prefs=runtime.tools.xtensa-esp32s2-elf-gcc.path=/Users/dirk/Library/Arduino15/packages/esp32/tools/xtensa-esp32s2-elf-gcc/esp-2021r2-patch5-8.4.0 -prefs=runtime.tools.xtensa-esp32s2-elf-gcc-esp-2021r2-patch5-8.4.0.path=/Users/dirk/Library/Arduino15/packages/esp32/tools/xtensa-esp32s2-elf-gcc/esp-2021r2-patch5-8.4.0 -prefs=runtime.tools.xtensa-esp-elf-gdb.path=/Users/dirk/Library/Arduino15/packages/esp32/tools/xtensa-esp-elf-gdb/11.2_20220823 -prefs=runtime.tools.xtensa-esp-elf-gdb-11.2_20220823.path=/Users/dirk/Library/Arduino15/packages/esp32/tools/xtensa-esp-elf-gdb/11.2_20220823 -prefs=runtime.tools.esptool_py.path=/Users/dirk/Library/Arduino15/packages/esp32/tools/esptool_py/4.5.1 -prefs=runtime.tools.esptool_py-4.5.1.path=/Users/dirk/Library/Arduino15/packages/esp32/tools/esptool_py/4.5.1 -prefs=runtime.tools.mkspiffs.path=/Users/dirk/Library/Arduino15/packages/esp32/tools/mkspiffs/0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=/Users/dirk/Library/Arduino15/packages/esp32/tools/mkspiffs/0.2.3 -verbose /Users/dirk/Documents/GitHub/FreeVarioGauge/src/FreeVarioGauge/FreeVarioGauge.ino
/Applications/Arduino.app/Contents/Java/arduino-builder -compile -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/dirk/Library/Arduino15/packages -hardware /Users/dirk/Documents/Arduino/hardware -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/dirk/Library/Arduino15/packages -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/dirk/Documents/Arduino/libraries -fqbn=esp32:esp32:esp32doit-devkit-v1:FlashFreq=80,UploadSpeed=115200,DebugLevel=none,EraseFlash=none -vid-pid=0000_0000 -ide-version=10813 -build-path /var/folders/6t/qjgfw2413f7ddnjzk_08btxw0000gp/T/arduino_build_710579 -warnings=none -build-cache /var/folders/6t/qjgfw2413f7ddnjzk_08btxw0000gp/T/arduino_cache_690012 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=/Users/dirk/Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc-esp-2021r2-patch5-8.4.0.path=/Users/dirk/Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0 -prefs=runtime.tools.riscv32-esp-elf-gcc.path=/Users/dirk/Library/Arduino15/packages/esp32/tools/riscv32-esp-elf-gcc/esp-2021r2-patch5-8.4.0 -prefs=runtime. [... cut ...] edtls/include -I/Users/dirk/Library/Arduino15/packages/esp32/hardware/esp32/2.0.11/tools/sdk/esp32/include/mbedtls/esp_crt_bundle/include -I/Users/dirk/Library/Arduino15/packages/esp32/hardware/esp32/2.0.11/tools/sdk/esp32/include/app_update/include -I/Users/dirk/Library/Arduino15/packages/esp32/hardware/esp32/2.0.11/tools/sdk/esp32/include/spi_flash/include -I/Users/dirk/Library/Arduino15/packages/esp32/hardware/esp32/2.0.11/tools/sdk/esp32/include/bootloader_support/include -I/Users/dirk/Libra

animevietsub, 2023-08-04:
Try to copy all of the Output **without** "Enable show verbose output during compile", because it's too long to send through Git Hub.

Blaubart, 2023-08-04:
Hope it is better:

Arduino15/packages/esp32/hardware/esp32/2.0.11/tools/sdk/esp32/include/perfmon/include -I/Users/dirk/Library/Arduino15/packages/esp32/hardware/esp32/2.0.11/tools/sdk/esp32/include/spiffs/include -I/Users/dirk/Library/Arduino15/packages/esp32/hardware/esp32/2.0.11/tools/sdk/esp32/include/ulp/include -I/Users/dirk/Library/Arduino15/packages/esp32/hardware/esp32/2.0.11/tools/sdk/esp32/include/wifi_provisioning/include -I/Users/dirk/Library/Arduino15/packages/esp32/hardware/esp32/2.0.11/tools/sdk/esp32/include/rmaker_common/include -I/Users/dirk/Library/Arduino15/packages/esp32/hardware/esp32/2.0.11/tools/sdk/esp32/include/esp_diagnostics/include -I/Users/dirk/Library/Arduino15/packages/esp32/hardware/esp32/2.0.11/tools/sdk/esp32/include/rtc_store/include -I/Users/dirk/Library/Arduino15/packages/esp32/hardware/esp32/2.0.11/tools/sdk/esp32/include/esp_insights/include -I/Users/dirk/Library/Arduino15/packages/esp32/hardware/esp32/2.0.11/tools/sdk/esp32/include/json_parser/upstream/include -I/Users/dirk/Library/Arduino15/packages/esp32/hardware/esp32/2.0.11/tools/sdk/esp32/include/json_parser/upstream -I/Users/dirk/Library/Arduino15/packages/esp32/hardware/esp32/2.0.11/tools/sdk/esp32/include/json_generator/upstream -I/Users/dirk/Library/Arduino15/packages/esp32/hardware/esp32/2.0.11/tools/sdk/esp32/include/esp_schedule/include -I/Users/dirk/Library/Arduino15/packages/esp32/hardware/esp32/2.0.11/tools/sdk/esp32/include/espressif__esp_secure_cert_mgr/include -I/Users/dirk/Library/Arduino15/packages/esp32/hardware/esp32/2.0.11/tools/sdk/esp32/include/esp_rainmaker/include -I/Users/dirk/Library/Arduino15/packages/esp32/hardware/esp32/2.0.11/tools/sdk/esp32/include/gpio_button/button/include -I/Users/dirk/Library/Arduino15/packages/esp32/hardware/esp32/2.0.11/tools/sdk/esp32/include/qrcode/include -I/Users/dirk/Library/Arduino15/packages/esp32/hardware/esp32/2.0.11/tools/sdk/esp32/include/ws2812_led -I/Users/dirk/Library/Arduino15/packages/esp32/hardware/esp32/2.0.11/tools/sdk/esp32/include/esp_littlefs/include -I/Users/dirk/Library/Arduino15/packages/esp32/hardware/esp32/2.0.11/tools/sdk/esp32/include/esp-dl/include -I/Users/dirk/Library/Arduino15/packages/esp32/hardware/esp32/2.0.11/tools/sdk/esp32/include/esp-dl/include/tool -I/Users/dirk/Library/Arduino15/packages/esp32/hardware/esp32/2.0.11/tools/sdk/esp32/include/esp-dl/include/typedef -I/Users/dirk/Library/Arduino15/packages/esp32/hardware/esp32/2.0.11/tools/sdk/esp32/include/esp-dl/include/image -I/Users/dirk/Library/Arduino15/packages/esp32/hardware/esp32/2.0.11/tools/sdk/esp32/include/esp-dl/include/math -I/Users/dirk/Library/Arduino15/packages/esp32/hardware/esp32/2.0.11/tools/sdk/esp32/include/esp-dl/include/nn -I/Users/dirk/Library/Arduino15/packages/esp32/hardware/esp32/2.0.11/tools/sdk/esp32/include/esp-dl/include/layer -I/Users/dirk/Library/Arduino15/packages/esp32/hardware/esp32/2.0.11/tools/sdk/esp32/include/esp-dl/include/detect -I/Users/dirk/Library/Arduino15/packages/esp32/hardware/esp32/2.0.11/tools/sdk/esp32/include/esp-dl/include/model_zoo -I/Users/dirk/Library/Arduino15/packages/esp32/hardware/esp32/2.0.11/tools/sdk/esp32/include/esp32-camera/driver/include -I/Users/dirk/Library/Arduino15/packages/esp32/hardware/esp32/2.0.11/tools/sdk/esp32/include/esp32-camera/conversions/include -I/Users/dirk/Library/Arduino15/packages/esp32/hardware/esp32/2.0.11/tools/sdk/esp32/include/espressif__esp-dsp/modules/dotprod/include -I/Users/dirk/Library/Arduino15/packages/esp32/hardware/esp32/2.0.11/tools/sdk/esp32/include/espressif__esp-dsp/modules/support/include -I/Users/dirk/Library/Arduino15/packages/esp32/hardware/esp32/2.0.11/tools/sdk/esp32/include/espressif__esp-dsp/modules/support/mem/include -I/Users/dirk/Library/Arduino15/packages/esp32/hardware/esp32/2.0.11/tools/sdk/esp32/include/espressif__esp-dsp/modules/windows/include -I/Users/dirk/Library/Arduino15/packages/esp32/hardware/esp32/2.0.11/tools/sdk/esp32/include/espressif__esp-dsp/modules/windows/hann/include -I/Users/dirk/Library/Arduino15/packages/esp32/hardware/esp32/2.0.11/tools/sdk/esp32/include/espressif__esp-dsp/modules/windows/blackman/include -I/Users/dirk/Library/Arduino15/packages/esp32/hardware/esp32/2.0.11/tools/sdk/esp32/include/espressif__esp-dsp/modules/windows/blackman_harris/include -I/Users/dirk/Library/Arduino15/packages/esp32/hardware/esp32/2.0.11/tools/sdk/esp32/include/espressif__esp-dsp/modules/windows/blackman_nuttall/include -I/Users/dirk/Library/Arduino15/packages/esp32/hardware/esp32/2.0.11/tools/sdk/esp32/include/espressif__esp-dsp/modules/windows/nuttall/include -I/Users/dirk/Library/Arduino15/packages/esp32/hardware/esp32/2.0.11/tools/sdk/esp32/include/espressif__esp-dsp/modules/windows/flat_top/include -I/Users/dirk/Library/Arduino15/packages/esp32/hardware/esp32/2.0.11/tools/sdk/esp32/include/espressif__esp-dsp/modules/iir/include -I/Users/dirk/Library/Arduino15/packages/esp32/hardware/esp32/2.0.11/tools/sdk/esp32/include/espressif__esp-dsp/modules/fir/include -I/Users/dirk/Library/Arduino15/packages/esp32/hardware/esp32/2.0.11/tools/sdk/esp32/include/espressif__esp-dsp/modules/math/include -I/Users/dirk/Library/Arduino15/packages/esp32/hardware/esp32/2.0.11/tools/sdk/esp32/include/espressif__esp-dsp/modules/math/add/include -I/Users/dirk/Library/Arduino15/packages/esp32/hardware/esp32/2.0.11/tools/sdk/esp32/include/espressif__esp-dsp/modules/math/sub/include -I/Users/dirk/Library/Arduino15/packages/esp32/hardware/esp32/2.0.11/tools/s [... cut ...]
Bibliothek FS in Version 2.0.0 im Ordner: /Users/dirk/Library/Arduino15/packages/esp32/hardware/esp32/2.0.11/libraries/FS  wird verwendet
Bibliothek SPIFFS in Version 2.0.0 im Ordner: /Users/dirk/Library/Arduino15/packages/esp32/hardware/esp32/2.0.11/libraries/SPIFFS  wird verwendet
Bibliothek Preferences in Version 2.0.0 im Ordner: /Users/dirk/Library/Arduino15/packages/esp32/hardware/esp32/2.0.11/libraries/Preferences  wird verwendet
Bibliothek ESP32Encoder in Version 0.10.1 im Ordner: /Users/dirk/Documents/Arduino/libraries/ESP32Encoder  wird verwendet
exit status 1
cannot convert 'bool' to 'puType' in assignment

animevietsub, 2023-08-04:
Because the library "ESP32Encoder" has been updated so you need to change the line in line **258**
`ESP32Encoder::useInternalWeakPullResistors = true;`
to 
`ESP32Encoder::useInternalWeakPullResistors = DOWN;`

Blaubart, 2023-08-05:
Thanks, it works!! But i have and other problem, too.
My ESP32 crashes because of this line:
Vario_Enc.setCount(16380);

animevietsub, 2023-08-05:
If you are compiling `FreeVarioGauge` projects make sure that `Vario_Enc.attachHalfQuad` is called before `Vario_Enc.setCount` in `FreeVarioGauge` projects that should be: 
```
  Vario_Enc.attachHalfQuad(23, 32);
  Vario_Enc.setCount(16380);
```
I see in the `FreeVarioGauge` projects your also need to use [**ESP32FS**](https://github.com/me-no-dev/arduino-esp32fs-plugin/releases/download/1.1/ESP32FS-1.1.zip) tools to upload these `micross` fonts files to SPIFFS, make sure you have that tools or it will return another error. 
After uploading `micross` fonts it should work fine. 

Check the [**building-installation-instructions_EN.pdf**](https://github.com/freevariode/FreeVarioGauge/blob/master/manuals/Changing-the-voltage-converter_EN.pdf) for more information.

Blaubart, 2023-08-05:
Thanks a lot!!!! It works!!

closed 2023-08-05 by madhephaestus