madhephaestus/ESP32Encoder #101: cannot even compile example for regular esp32 dev board

42loop, 2024-02-24:
log:

Arduino: 1.8.16 (Linux), TD: 1.57, Board: "ESP32 Dev Module, Disabled, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 115200, Core 1, Core 1, None, Disabled, Disabled"

... [cut for readability, full log attached
[build_log.txt](https://github.com/madhephaestus/ESP32Encoder/files/14392541/build_log.txt)
]

/home/ulf/sketchbook/libraries/ESP32Encoder/src/ESP32Encoder.cpp:18:8: error: 'portMUX_TYPE' does not name a type
   18 | static portMUX_TYPE spinlock = portMUX_INITIALIZER_UNLOCKED;
      |        ^~~~~~~~~~~~
/home/ulf/sketchbook/libraries/ESP32Encoder/src/ESP32Encoder.cpp: In function 'void esp32encoder_pcnt_intr_handler(void*)':
/home/ulf/sketchbook/libraries/ESP32Encoder/src/ESP32Encoder.cpp:19:52: error: 'spinlock' was not declared in this scope
   19 | #define _ENTER_CRITICAL() portENTER_CRITICAL_SAFE(&spinlock)
      |                                                    ^~~~~~~~
/home/ulf/sketchbook/libraries/ESP32Encoder/src/ESP32Encoder.cpp:75:9: note: in expansion of macro '_ENTER_CRITICAL'
   75 |         _ENTER_CRITICAL();
      |         ^~~~~~~~~~~~~~~
/home/ulf/sketchbook/libraries/ESP32Encoder/src/ESP32Encoder.cpp:19:27: error: 'portENTER_CRITICAL_SAFE' was not declared in this scope
   19 | #define _ENTER_CRITICAL() portENTER_CRITICAL_SAFE(&spinlock)
      |                           ^~~~~~~~~~~~~~~~~~~~~~~
/home/ulf/sketchbook/libraries/ESP32Encoder/src/ESP32Encoder.cpp:75:9: note: in expansion of macro '_ENTER_CRITICAL'
   75 |         _ENTER_CRITICAL();
      |         ^~~~~~~~~~~~~~~
/home/ulf/sketchbook/libraries/ESP32Encoder/src/ESP32Encoder.cpp:20:26: error: 'portEXIT_CRITICAL_SAFE' was not declared in this scope
   20 | #define _EXIT_CRITICAL() portEXIT_CRITICAL_SAFE(&spinlock)
      |                          ^~~~~~~~~~~~~~~~~~~~~~
/home/ulf/sketchbook/libraries/ESP32Encoder/src/ESP32Encoder.cpp:95:9: note: in expansion of macro '_EXIT_CRITICAL'
   95 |         _EXIT_CRITICAL();
      |         ^~~~~~~~~~~~~~
/home/ulf/sketchbook/libraries/ESP32Encoder/src/ESP32Encoder.cpp: At global scope:
/home/ulf/sketchbook/libraries/ESP32Encoder/src/ESP32Encoder.cpp:112:8: error: 'IRAM_ATTR' does not name a type
  112 | static IRAM_ATTR void ipc_install_isr_on_core(void *arg) {
      |        ^~~~~~~~~
/home/ulf/sketchbook/libraries/ESP32Encoder/src/ESP32Encoder.cpp: In member function 'void ESP32Encoder::attach(int, int, encType)':
/home/ulf/sketchbook/libraries/ESP32Encoder/src/ESP32Encoder.cpp:140:9: error: 'gpio_pad_select_gpio' was not declared in this scope; did you mean 'esp_rom_gpio_pad_select_gpio'?
  140 |         gpio_pad_select_gpio(aPinNumber);
      |         ^~~~~~~~~~~~~~~~~~~~
      |         esp_rom_gpio_pad_select_gpio
/home/ulf/sketchbook/libraries/ESP32Encoder/src/ESP32Encoder.cpp:203:87: error: 'xPortGetCoreID' was not declared in this scope
  203 |                 if (isrServiceCpuCore == ISR_CORE_USE_DEFAULT || isrServiceCpuCore == xPortGetCoreID()) {
      |                                                                                       ^~~~~~~~~~~~~~
/home/ulf/sketchbook/libraries/ESP32Encoder/src/ESP32Encoder.cpp:210:81: error: 'ipc_install_isr_on_core' was not declared in this scope
  210 |                         esp_err_t er = esp_ipc_call_blocking(isrServiceCpuCore, ipc_install_isr_on_core, &ipc_ret_code);
      |                                                                                 ^~~~~~~~~~~~~~~~~~~~~~~
/home/ulf/sketchbook/libraries/ESP32Encoder/src/ESP32Encoder.cpp: In member function 'void ESP32Encoder::setCount(int64_t)':
/home/ulf/sketchbook/libraries/ESP32Encoder/src/ESP32Encoder.cpp:19:52: error: 'spinlock' was not declared in this scope
   19 | #define _ENTER_CRITICAL() portENTER_CRITICAL_SAFE(&spinlock)
      |                                                    ^~~~~~~~
/home/ulf/sketchbook/libraries/ESP32Encoder/src/ESP32Encoder.cpp:253:9: note: in expansion of macro '_ENTER_CRITICAL'
  253 |         _ENTER_CRITICAL();
      |         ^~~~~~~~~~~~~~~
/home/ulf/sketchbook/libraries/ESP32Encoder/src/ESP32Encoder.cpp:19:27: error: 'portENTER_CRITICAL_SAFE' was not declared in this scope
   19 | #define _ENTER_CRITICAL() portENTER_CRITICAL_SAFE(&spinlock)
      |                           ^~~~~~~~~~~~~~~~~~~~~~~
/home/ulf/sketchbook/libraries/ESP32Encoder/src/ESP32Encoder.cpp:253:9: note: in expansion of macro '_ENTER_CRITICAL'
  253 |         _ENTER_CRITICAL();
      |         ^~~~~~~~~~~~~~~
/home/ulf/sketchbook/libraries/ESP32Encoder/src/ESP32Encoder.cpp:20:26: error: 'portEXIT_CRITICAL_SAFE' was not declared in this scope
   20 | #define _EXIT_CRITICAL() portEXIT_CRITICAL_SAFE(&spinlock)
      |                          ^~~~~~~~~~~~~~~~~~~~~~
/home/ulf/sketchbook/libraries/ESP32Encoder/src/ESP32Encoder.cpp:255:9: note: in expansion of macro '_EXIT_CRITICAL'
  255 |         _EXIT_CRITICAL();
      |         ^~~~~~~~~~~~~~
/home/ulf/sketchbook/libraries/ESP32Encoder/src/ESP32Encoder.cpp: In member function 'int64_t ESP32Encoder::getCountRaw()':
/home/ulf/sketchbook/libraries/ESP32Encoder/src/ESP32Encoder.cpp:19:52: error: 'spinlock' was not declared in this scope
   19 | #define _ENTER_CRITICAL() portENTER_CRITICAL_SAFE(&spinlock)
      |                                                    ^~~~~~~~
/home/ulf/sketchbook/libraries/ESP32Encoder/src/ESP32Encoder.cpp:261:9: note: in expansion of macro '_ENTER_CRITICAL'
  261 |         _ENTER_CRITICAL();
[... 29 lines cut ...]
/home/ulf/sketchbook/libraries/ESP32Encoder/src/ESP32Encoder.cpp:280:9: note: in expansion of macro '_EXIT_CRITICAL'
  280 |         _EXIT_CRITICAL();
      |         ^~~~~~~~~~~~~~
/home/ulf/sketchbook/libraries/ESP32Encoder/src/ESP32Encoder.cpp: In member function 'int64_t ESP32Encoder::clearCount()':
/home/ulf/sketchbook/libraries/ESP32Encoder/src/ESP32Encoder.cpp:19:52: error: 'spinlock' was not declared in this scope
   19 | #define _ENTER_CRITICAL() portENTER_CRITICAL_SAFE(&spinlock)
      |                                                    ^~~~~~~~
/home/ulf/sketchbook/libraries/ESP32Encoder/src/ESP32Encoder.cpp:285:9: note: in expansion of macro '_ENTER_CRITICAL'
  285 |         _ENTER_CRITICAL();
      |         ^~~~~~~~~~~~~~~
/home/ulf/sketchbook/libraries/ESP32Encoder/src/ESP32Encoder.cpp:19:27: error: 'portENTER_CRITICAL_SAFE' was not declared in this scope
   19 | #define _ENTER_CRITICAL() portENTER_CRITICAL_SAFE(&spinlock)
      |                           ^~~~~~~~~~~~~~~~~~~~~~~
/home/ulf/sketchbook/libraries/ESP32Encoder/src/ESP32Encoder.cpp:285:9: note: in expansion of macro '_ENTER_CRITICAL'
  285 |         _ENTER_CRITICAL();
      |         ^~~~~~~~~~~~~~~
/home/ulf/sketchbook/libraries/ESP32Encoder/src/ESP32Encoder.cpp:20:26: error: 'portEXIT_CRITICAL_SAFE' was not declared in this scope
   20 | #define _EXIT_CRITICAL() portEXIT_CRITICAL_SAFE(&spinlock)
      |                          ^~~~~~~~~~~~~~~~~~~~~~
/home/ulf/sketchbook/libraries/ESP32Encoder/src/ESP32Encoder.cpp:287:9: note: in expansion of macro '_EXIT_CRITICAL'
  287 |         _EXIT_CRITICAL();
      |         ^~~~~~~~~~~~~~
Using library ESP32Encoder at version 0.10.2 in folder: /home/ulf/sketchbook/libraries/ESP32Encoder 
exit status 1
Error compiling for board ESP32 Dev Module.

42loop, 2024-02-27:
i was using espressif systems 3.0.0-alpha3,
but downgrading to 2.0.11 did the trick.

should i close the issue ?

madhephaestus, 2024-02-29:
Can you change the title to reflect that this is a 3.x.x compilation issue?

tueddy, 2024-03-01:
I can reproduce the error with PlatformIO, using ESP-IDF 4.x your library works fine but with Arduino 3 (ESP-IDF 5.1) throws compiler errors above.

platform.ini using IDF 5.1:

platform = espressif32
framework = arduino
platform_packages =
	  platformio/framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#3.0.0-alpha3
	  platformio/framework-arduinoespressif32-libs @ https://github.com/espressif/esp32-arduino-libs.git#idf-release/v5.1

tueddy, 2024-03-01:
This solves the compile error (ESP32Encoder.cpp):

`#include <freertos/FreeRTOS.h>`
`#include <rom/gpio.h>`

After compile you get some warn messages:
` warning: #warning "legacy pcnt driver is deprecated, please migrate to use driver/pulse_cnt.h" [-Wcpp]`
and
`ESP32Encoder.cpp:84:33: warning: compound assignment with 'volatile'-qualified left operand is deprecated [-Wvolatile]`

closed 2024-03-05 by madhephaestus

jschoch, 2024-03-11:
This doesn't seem to work for platformio.

```
#include "esp_log.h"
#include "esp_ipc.h"
#if ( defined(ESP_ARDUINO_VERSION_MAJOR) && (ESP_ARDUINO_VERSION_MAJOR >= 3) )
	#include <freertos/FreeRTOS.h>
	#include <rom/gpio.h>
#else
	#warning "the detection failed"
#endif

```

```
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (2023.10.12+sha.d076cd6) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 3.0.0
 - tool-cmake @ 3.21.3
 - tool-esptoolpy @ 1.40700.0+dev2 (4.7.0)
 - tool-idf @ 1.0.1
 - tool-mconf @ 1.4060000.20190628 (406.0.0)
 - tool-mklittlefs @ 3.2.0
 - tool-mkspiffs @ 2.230.0 (2.30)
 - tool-ninja @ 1.9.0
 - tool-riscv32-esp-elf-gdb @ 11.2.0+20220823
 - tool-xtensa-esp-elf-gdb @ 11.2.0+20230208
 - toolchain-esp32ulp @ 1.23500.220830 (2.35.0)
 - toolchain-xtensa-esp32 @ 12.2.0+20230208
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 24 compatible libraries
Scanning dependencies...
Dependency Graph
|-- ESP32Encoder @ 0.11.2
|-- Ticker @ 2.0.0
Building in release mode
Compiling .pio\build\esp32dev\src\main.cpp.o
Building .pio\build\esp32dev\bootloader.bin
Generating partitions .pio\build\esp32dev\partitions.bin
Compiling .pio\build\esp32dev\lib193\ESP32Encoder\ESP32Encoder.cpp.o
esptool.py v4.7-dev
Creating esp32 image...
Merged 1 ELF section
Successfully created esp32 image.
Compiling .pio\build\esp32dev\lib193\ESP32Encoder\InterruptEncoder.cpp.o
Compiling .pio\build\esp32dev\lib448\Ticker\Ticker.cpp.o
Compiling .pio\build\esp32dev\FrameworkArduino\Esp.cpp.o
Compiling .pio\build\esp32dev\FrameworkArduino\FirmwareMSC.cpp.o
Compiling .pio\build\esp32dev\FrameworkArduino\FunctionalInterrupt.cpp.o
In file included from .pio/libdeps/esp32dev/ESP32Encoder/src/ESP32Encoder.h:3,
                 from .pio/libdeps/esp32dev/ESP32Encoder/src/ESP32Encoder.cpp:8:
C:/Users/jesse/.platformio/packages/framework-arduinoespressif32@src-9278b63539f5b4fad08d9395849529be/tools/esp32-arduino-libs/esp32/include/driver/deprecated/driver/pcnt.h:15:2: warning: #warning "legacy pcnt driver is deprecated, please migrate to use driver/pulse_cnt.h" [-Wcpp]
   15 | #warning "legacy pcnt driver is deprecated, please migrate to use driver/pulse_cnt.h"
      |  ^~~~~~~
.pio/libdeps/esp32dev/ESP32Encoder/src/ESP32Encoder.cpp:19:10: warning: #warning "the detection failed" [-Wcpp]
   19 |         #warning "the detection failed"

... lots of errors
```

tueddy, 2024-03-11:
You're right, it does not compile with Arduino 3 / ESP-IDF 5. 
With current define we need additionally
`#include <Arduino.h>`

or change the compiler condition to
`#if ESP_IDF_VERSION_MAJOR >= 5`