madhephaestus/ESP32Encoder #104: Encoder limit? ESP32S3 Bootloop Issue "Too Many Encoders!"

zachtsen55, 2024-03-05:
I have a sketch with two dual rotary encoders that was doing fine up until I added one more single encoder, and the ESP32 would just bootloop with a variety of different outputs depending on how I change the sketch, but it always returns the following:

-> E (7) ESP32Encoder: Too many encoders, FAIL!

Is there a limit to the amount of encoders an ESP32S3 can take? I can't seem to find anything on this issue and supposedly all GPIOs are interrupt capable. My total encoders are 5, which doesn't seem like many but I could be wrong, can anyone advise? Thanks!

madhephaestus, 2024-03-05:
That is governed by `PCNT_UNIT_MAX` which is different for each core. The original ESP32 had 10 channels, i guess the s3 has just 2. That is a bit sad that they reduced the functionality so much between cores. I still find the original ESP32 to be  the best core for general use. I would only use the reduced functionality cores if i needed something specific added like the USB host features, and could make do with significantly reduce peripherals.

closed 2024-03-05 by madhephaestus

madhephaestus, 2024-03-05:
You can read about the PCNT module that is built into the cores here. Since the same module is used for all the cores, the documentation is generic and does not specify the differing number of pcnt channels per core. 

https://docs.espressif.com/projects/esp-idf/en/stable/esp32s3/api-reference/peripherals/pcnt.html#pcnt-resource-allocation

You might want to take a look at InterruptEncoder objects to complete your project without having to change hardware. Beware: InteruptEncoder can choke out your core is the encoder ticks too fast. You can have as many as you want, but you need to make sure that the number of ticks per second is low enough for the CPU to handle all of the encoding.

zachtsen55, 2024-03-05:
Thanks so much for the reply! Really appreciate it :)
I'll definitely read up on PCNT modules and their limitations, that's a real shame, but thanks for the suggestion for looking at interrupter encoder objects, will do some tests and see if that works, since all my encoders are only for inputting data by hand, I highly doubt I'll be maxing out any ticks-per-second limits I suspect. Again thanks for the quick response!

zachtsen55, 2024-03-06:
I tried to use a different method of adding encoders with attachInterrupt, I think it's a hardware limitation of the ESP32S3 unfortunately, I ordered a WROOM N16R8, hopefully that'll solve everything!