madhephaestus/ESP32Encoder #109: wrong counting with fast encoder

tappete, 2024-04-12:
Your library is truly fantastic, and I've managed to use a series of rotary encoders.

However, now I'm encountering an issue with a very fast 16-bit encoder.

The transitions between A and B forms are 1 microsecond apart when I move quickly (I've analyzed the waveforms), and the count is completely wrong.

However, when the encoder rotates slowly, I don't lose steps, so I believe everything is okay at the circuit level.
Could this be related to the ISR issue, and if so, how can I configure and conduct tests?
Or could it be something else like interrupts?

madhephaestus, 2024-04-12:
That issue is likely caused by the filter kicking in and trying to prevent erronious counts when signal noise exists on the lines. Try setting encoder.setFilter(0); this will disable the pcnt filter. If that helps, it was your issue. If it doesnt help, then i guess you are running faster than the device can detect :/

tappete, 2024-04-15:
Great, problem solved, now it counts without any issues perfectly.

Thank you very much!

Dave

closed 2024-04-15 by tappete