madhephaestus/ESP32Encoder #115: compatibility with ESP32 core V3

0x0fe, 2024-07-25:
With the ESP32 Core V3 some types and macros are now invalid and block compilation.

```
'portMUX_TYPE' does not name a type
   15 | static portMUX_TYPE spinlock = portMUX_INITIALIZER_UNLOCKED;
   
   \ESP32Encoder.cpp:16:52: error: 'spinlock' was not declared in this scope
   16 | #define _ENTER_CRITICAL() portENTER_CRITICAL_SAFE(&spinlock)
```

0x0fe, 2024-07-25:
my bad, it was my version which was old (0.9) and for some reason the IDE did not allo to update. I had to remove it and reinstall to be able to get the last revision, now there is no more error.

closed 2024-07-25 by 0x0fe

hippo5329, 2024-07-26:
You may follow my wiki to use PlatformIO/VSCode. There will be less issues.
https://github.com/hippo5329/micro_ros_arduino_examples_platformio/wiki