madhephaestus/ESP32Encoder #117: legacy pcnt driver is deprecated

farTooOld, 2024-09-14:
#warning "legacy pcnt driver is deprecated, please migrate to use driver/pulse_cnt.h"

Arduino IDE 2.3.2
#include <ESP32Encoder.h>

Using Two Encoders 600ppr. The Example codes give the same Warning. But the Encoders seem to be working (except of dropping some pulses when going from Positive direction to Negative direction.

I've searched and searched for solution but haven't found any solution. Is something Not working? How to stop the Warning if noting is wrong?

What do I need to do (and how to do it)?

(otherwise, it's the Best Encoder lib).

Thank you

anthonyjclark, 2024-10-02:
Here is some information on the change and the migration:
https://docs.espressif.com/projects/esp-idf/en/stable/esp32/migration-guides/release-5.x/5.0/peripherals.html#pulse-counter-driver

I'll try to submit a PR if I have the time, but I do not expect that I will anytime soon.

whogarden, 2024-12-10:
It's just a warning. Question : can we use V0.11.7 with no pb?