madhephaestus/ESP32Encoder #118: How to setup range up/down for each encoder

gavron04, 2024-11-12:
Hello. How to add up and down range for each encoder? Like moving between 0 to 10 or different value? ESP32, 

For example. Range between 0 and 3 (4 positions 0;1;2;3)
```
sel = encoder2.getCount();
if(sel>3)
{
  encoder2.pauseCount();
}
```
encoder go to 4 and then stop. Not 3 and when moving up for 4 - is paused. 

I cant find easy solution in .cpp or .h for this.

Thanks for help

gavron04, 2024-11-13:
or how to write function if direction of rotate was for left or right side? (up/down)

madhephaestus, 2024-11-13:
There is not API for this, but you can try to modify the values here: https://github.com/madhephaestus/ESP32Encoder/blob/master/src/ESP32Encoder.cpp#L185 

If modifying those values is working for you, you could add class variables for those values with the _INT16_MAX and _INT16_MIN as the defaults, and an API to set them before attach() is called. I would be happy to accept a PR addin that functionality once you have tested it working for your use case.

gavron04, 2024-11-13:
But it is for all connected encoders, not for each encoder, right? I need separate range for 2 encoders, for example:

one for 0-120, second for 0-10

madhephaestus, 2024-11-13:
that's why i said "class variable" not "static variable" ;) If you made it a static variable it would be for all encoders, class variables are templated into each instance of an object.

gavron04, 2024-11-13:
OK. How about:
`	r_enc_config.lctrl_mode = PCNT_MODE_KEEP;    // Rising A on HIGH B = CW Step
	r_enc_config.hctrl_mode = PCNT_MODE_REVERSE; // Rising A on LOW B = CCW Step`
	
	there is no possibly to use it as detect if encoder has CW or CCW step?

gavron04, 2024-11-13:
Maybe it will help :)

```
  newvalue = encoder.getCount();
  if (newvalue > oldvalue) {
    oldvalue = newvalue;
    Serial.println("increase");
  }

  if (newvalue < oldvalue) {
    oldvalue = newvalue;
    Serial.println("decrease");
  }

```